InfiniTimeOrg/InfiniTime #183: heart rate sensor: measurement is stopping, when the screen is going off

PetersOtto, 2021-01-26:
I installed v 0.11.0.
The heart rate sensor works great! 

I compare the measurement of the PineTime with the measurement of my Garmin Forerunner. The value are in the same range. Thats  good, I thing.

However the measurement is stopping, when the screen is going off.
It would be great to have a button, that deactivate the power off-mode from the screen. Then it would be possible to have a longer view on the measurement.

It seems that the Heart Rate Sensor of the Garmin is connected to accelerator- / motion sensor.  The Heart Rate Sensor starts working, when movement is detected. Perhaps that could a solution in future.

Thank you for working on this firmware!

JF002, 2021-01-27:
Thanks for the comparison with another device! Good to know the results seem correct!

The behavior you describe is expected : this is a first integration of the heart rate sensor, and the UI/UX can still be improved in further release. 

I cannot measure it for now, but I guess that leaving the heart rate sensor all the time might use a lot of power and reduce the battery life. As you said, using info from the motion sensor might be a good idea, but this sensor is not integrated yet in InfiniTime. There's still work to do ;)

> Thank you for working on this firmware!

Thanks, I appreciate! And thanks for using it and sharing your experience!

piggz, 2021-02-04:
Id like to see a couple of modes, each should save power in different ways:
1. Allow the user to turn on/off the HRM explicitly.  This would be on constantly until the user turns it off.  This would be useful for exercise, where for example the watch is supplying the heartrate to another device
2. Have the device turn on the HRM at a user specified interval, typically 60 seconds.  This would be useful for when the firmware has access to a filesystem for logging the heartrate.  A companion app can then download that data later.

JF002, 2021-02-04:
I also think of a 3rd case : in addition to turn the HRM ON at a specified interval, we could also use the motion sensor to turn the HRM ON on motion even.

... But first, we need to implement the motion sensor :)

jones139, 2021-02-06:
> ... But first, we need to implement the motion sensor :)

I know, I have been feeling bad about lack of progress.   I am encouraged though that this HRM implementation is using FreeRTOS tasks and a message queue as far as I can tell from a quick look so I will use that style to implement the accelerometer......but first I need to get the new version to compile so I can add in the accelerometer - it is failing at the moment saying it will not fit in the flash memory "ld: pinetime-app-0.12.0.out section `.ARM.exidx' will not fit in region `FLASH'", so I must have my configuration wrong.......

adamm, 2021-02-14:
> > ... But first, we need to implement the motion sensor :)

I thought I'd take a whack at adding in the BMA421 sensor support, using the heartrate controller/task/service as a template.   I figure just simply updating the stepCount on the clock is a good first milestone.  If it works, i'll create a new issue and PR.

danielbarry, 2021-03-04:
@JF002 Once the watch has been set to "start" mode in the heart rate app, I think it should not turn off the sensor updates, which would mean keeping the I2C bus on (although other devices on the bus can still be switched off).

I imagine it would be more power efficient to wake the I2C bus as it's needed, rather than to have it on all the time. I imagine you would want to potentially have at least two logging modes - background day-to-day HR monitor and exercise monitoring.

A while back I started implementing a power-management task that would shut down devices not in use and wake them up as required. For example, if there are no changes to the display and nothing being read/write, you could even turn off the SPI bus whilst the display does nothing to save power.

Dperrigo18, 2021-08-01:
>     1. Allow the user to turn on/off the HRM explicitly.  This would be on constantly until the user turns it off.  This would be useful for exercise, where for example the watch is supplying the heartrate to another device
I actually second this. I'm using [FitoTrack](https://codeberg.org/jannis/FitoTrack) off of FDroid and during my exercise I turned it on for a minute and it picked it up! 
I was pretty surprised. [Picture](https://imgur.com/a/xxvfadf)

JF002, 2021-08-01:
> I actually second this. I'm using FitoTrack off of FDroid and during my exercise I turned it on for a minute and it picked it up!
I was pretty surprised. Picture

Hey thanks for letting us know about FitoTrack, I didn't know this app! It's nice it's supporting the standard BLE endpoint for HR data!

ThePinkUnicorn6, 2021-08-15:
Hi, I am also using FitoTrack and can also say having the functionality to keep the heart rate monitor on with the screen off would be really usefull.

NCC1701M, 2022-04-10:
What's the state of the issue? Other smartwatches are able to track the heart rate all the time and still run for 3 or 4 days without charging. InfiniTime should be able to do the same.

JF002, 2022-04-10:
InfiniTime does not support continuous HR measurements and stops the sensor as soon as it goes to sleep (display off) for now. Of course, InfiniTime should be able to do the same as other smartwatches... All it takes is someone to implement it ;-)

tlhonmey, 2022-05-16:
As a work-around, depending on the kind of exercise, setting a sensitive shake to wake and a long screen timeout might keep it going for you.

dadall, 2022-06-07:
Plus choosing 15 seconds as screen timeout let the sensor enough time to get the HRM. It could be nice to add a 10 seconds timeout option, though.

6b6279, 2022-07-16:
15 seconds seems to be enough, but there is a further issue: The HR display on the home screen resets when the display is turned off. So if the HR is calculated thirteen seconds into the timeout, the measurement is discarded even if the user turns on the display a second after the timeout.

I guess the best option is to just remember the last heart rate measured and mark it as "stale" (maybe colour it yellow) instead of just showing 0. Of course none of this would be necessary if the HR sensor worked while the display is turned off.

patricgruber, 2023-08-26:
The issue of resetting the heart rate to 0 after a screen lock and that the HR measurement is stopped when the screen is turned off will be fixed with the PR https://github.com/InfiniTimeOrg/InfiniTime/pull/1718

florisre, 2024-07-04:
I think this was already implemented and currently is in real life use thanks to https://github.com/OpenSeizureDetector/PineTimeSD/pull/3 by @jones139. Maybe his previous work can be used in closing this or also in https://github.com/InfiniTimeOrg/InfiniTime/pull/1718?

mark9064, 2025-11-05:
Resolved by #2322 :)

closed 2025-11-05 by mark9064