InfiniTimeOrg/InfiniTime #185: Instability prior to validation

pjsfx, 2021-01-27:
Just installed 0.11. It took me about four attempts to get things going as every time I tried an app (tried draw, heart rate monitor, pong and even just the info tab) the watch would lock up and require a hard reset and re-download of the dfu.zip file. Eventually I just verified the firmware without doing any tests and now everything seems fine i.e. all those app 'work'. Just an information point for anyone who experiences the same.

Avamander, 2021-01-27:
Never do that, especially on a sealed device. If it's unstable, report it and wait for an update. It's simply not safe to confirm an unstable firmware on a sealed watch.

JF002, 2021-01-27:
I flashed 0.11 on my sealed unit just after the release and haven't noticed that kind of behaviors yet, but I'll definitely pay attention to any suspicious issue.
Validating the app only consists in setting a flag in flash memory to prevent the bootloader from reverting to the previous version on next reset so it should not have any effect on the functioning the firmware...

pjsfx, 2021-01-28:
> Never do that, especially on a sealed device. If it's unstable, report it and wait for an update. It's simply not safe to confirm an unstable firmware on a sealed watch.

Well, to be fair, it's not a sealed unit and both the watch face and menu grid were perfectly stable, so I had every expectation of being able to recover, but no, not a good idea on a sealed unit.

pjsfx, 2021-01-28:
> I flashed 0.11 on my sealed unit just after the release and haven't noticed that kind of behaviors yet, but I'll definitely pay attention to any suspicious issue.
> Validating the app only consists in [...]

Surely you also delete or overwrite the previous version?  I probably rebooted immediately after validation to see if the update had 'stuck'

petterhs, 2021-01-28:
The previous version is stored so you can revert manually to it.
This is done by pressing and holding the button while the green pine logo is loading. Release is when logo turns blue. If you have an emergency firmware loaded aswell you can hold the button further until it turns red.

JF002, 2021-02-16:
> The previous version is stored so you can revert manually to it.
> This is done by pressing and holding the button while the green pine logo is loading. Release is when logo turns blue. If you have an emergency firmware loaded aswell you can hold the button further until it turns red.

This is only valid with the new beta bootloader. The original bootloader also provides a similar way to revert (long push when the boot logo is displayed) but does not provide any feedback on the UI, if I recall correctly. The original bootloader does not support the recovery firmware, though.

JF002, 2021-02-16:
@pjsfx Any news on this issue? Does it still happen on newer versions? Do you have any other unit to test and see if you can reproduce this issue on another device?

pjsfx, 2021-02-17:
Had the same issue with 0.12.1, but 0.13.0 worked fine.

closed 2021-02-18 by pjsfx