InfiniTimeOrg/InfiniTime #200: Time doesn't sync with nRFConnect 2.4.7

gillham, 2021-02-05:
The iOS app store version of nRFConnect (2.4.7 build 70) doesn't sync time with the PineTime running InfiniTime.

The version of nRFConnect available on the app store doesn't match any of the screen shots so I'm assuming you're using an older version.

Using 'Peripheral' along the bottom and then 'Add Advertiser', I added a 'Current Time Service' advertiser.  I then turned it on via the slider.  (it doesn't stay on after the phone goes to sleep, but seems to stay on while actively using nRFConnect)
As far as I can tell this should be the equivalent of the 'GATT' server for 'Current Time Service' with the older app from your screenshots.

Unfortunately it doesn't seem to sync.  If you have any suggestions on how to debug this, I would appreciate it.

Avamander, 2021-02-05:
> The version of nRFConnect available on the app store doesn't match any of the screen shots so I'm assuming you're using an older version.

Nope, it's just the Android version.

I'm not aware of anyone figuring it out automatic time sync on iOS, but there have been attempts by others. If you find a solution, do share it.

gillham, 2021-02-05:
Thanks for the response.
I guess the docs are a little unclear since it mentions nRFConnect for iOS (which does work for DFU) and nRFConnect for time sync (but doesn't specifically mention iOS or Android, just "smartphone")
I'll try to send a pull request for a minor update to the getting started docs.

closed 2021-02-05 by gillham