InfiniTimeOrg/InfiniTime #201: Is it possible to power down / turn off the Pinetime?

gillham, 2021-02-05:
I haven't found any mention of shutting the watch off.  I have seen a few references to waiting until the battery runs dead, but I am assuming there should be some mechanism to actually power off?

Is this possible with the hardware / bootloader / firmware?  Simplest use case is something like "airplane mode" by just turning it off completely.

JF002, 2021-02-06:
We sometimes recommend to wait for a full battery drain in order to workaround a firmware/bootloader lock or freeze because that's the only way we have to trig a hardware reset of the MCU.

Pinetime does not have any hardware power button or reset button. The only reset we can issue are watchdog reset and software reset. Watchdog reset is automatically issued by the watchdog if the firmware crashes, but not when it's in bootloader mode. Software reset is trigged by the software when you push the button for 5-10s, but it does not work in bootloader mode or when the firmware is frozen.

So, no, unfortunately, there is no when to hardware reset your pinetime or power it off, this is by hardware design. You can do it manually by issuing a reset via the SWD interface or disconnecting/reconnecting the battery, assuming your unit is not a sealed device.

closed 2021-02-16 by JF002

blaueente, 2021-05-12:
I understand the "hardware reset" issue. But the question is also about software "off":
1. Is there any possibility to turn it "off", i.e. in a state where it deactivates anything it can, to save most power it can, such that it can only be woken by (long-)pressing the button, and it boots again? 
2. Also a 2nd mode of "airplaine" would be cool, such that it just disables Bluetooth, but the user can still use it to, e.g. read the current date and time.
For these use cases a software button would be enough.

Avamander, 2021-05-12:
@blaueente 
> Is there any possibility to turn it "off", i.e. in a state where it deactivates anything it can, to save most power it can, such that it can only be woken by (long-)pressing the button, and it boots again?

Some kind of storage mode is doable. Though it is a separate issue and should be discussed under that.

danielgjackson, 2022-01-18:
Has anyone got an existing plan for supporting an "off"/"shipping"/"storage" (very low power) mode?  How do Pine64 ship the devices while retaining battery charge?

Perhaps it might be best done in the bootloader but, failing that, I think it could be done as just a low-power delay in starting the application, waiting for a button press and/or to be (re-)connected to power:

1. Have a setting to represent that the device should be in shipping mode.  Perhaps as a "noinit" item in RAM (alternatively, a setting written to flash memory).  Long-press in the application can bring up a "power menu" (options for "Off", "Reboot", "Cancel").  If turning off, the setting is written (e.g. to noinit RAM) and the device restarted.

2. Rearrange peripheral code so that everything is set to its lowest power state at the start of main(), and before the task scheduler is started.  So, for example, the display should be off, SPI bus off, external memory off, motor off, accelerometer in its lowest power mode, heart rate monitor in standby.  The button and connected/charging status must be correctly configured as inputs.  I think there should also be some additional checking that this firmware build is definitely for the hardware its currently running on (i.e. all the peripherals behave as expected) -- otherwise the low-power loop cannot be exited if the button/charging state are not actually on the correct I/O pins!

3. If the "off" setting is enabled (and, perhaps, also if the battery is very low on power or it was detected that the device restarted from a brown-out condition?): 

    * clear the "off" setting, so that any subsequent restart will run the main app
    * adjust any remaining I/O pins and peripherals to lowest power settings
    * set up a low power and low frequency timer at `interval` (e.g. 500 msec)
    * `bool was_connected = true;`
    * `int button_count = 0;`
    * `uint64_t elapsed_ticks = 0;`
    * loop forever:
      * sleep the processor until a timer interrupt
      * `elapsed_ticks++;`
      * `connected = ` whether the device is charging/connected
      * `if (!was_connected && connected) break;` -- run application when transitioning from not-connected to connected (stay asleep when remaining connected from the start).
      * `was_connected = connected;`
      * if button pressed:
        * `button_count++;`
      * else:
        * reset the watchdog timer
        * `if (button_count >= 2) break;` -- run application if the button was held then released over a medium amount of time.  This allows a very long press to fire the watchdog interrupt to reset without starting the main app.
        * `button_count = 0;`
    * stop the low-power timer
    * update the system clock based on the missed time: `elapsed_ticks * interval`.
    * restore any needed I/O pins

...hopefully the main app can then just continue (alternatively, the device could just reset).