InfiniTimeOrg/InfiniTime #202: Infinitime does not display all of the notification data sent by Amazfish

willlovesbearz, 2021-02-09:
I noticed while looking at the debug output of `harbour-amazfishd` that the application is sending more data than the watch is displaying. everything referencing notifications in the output references 3 separate fields, `appName`, `summary`, and `body`. the first field, `appName`, contains the text the notification app shows.

Avamander, 2021-02-09:
As I mentioned in the chat, if it works with Gadgetbridge but not Amazfish, it's probably a misinterpretation of the protocol by Amazfish, thus its issue.

JF002, 2021-02-16:
We found the origin of this issue yesterday on the chat : Amazfish sends those 3 fields separated by a 0x00 in that order (app name, summary, body). 
Infinitime does not handle those 3 fields, it only expect 1 single message (as a null terminated string). When it receives the notification from Amazfish, the whole buffer is received, but only the first field (up to the 0x00 character) is displayed. 

I think InfiniTime can be easily modified to take these fields into account and display them in the Notification app (app name as the title of the notification, summary or body as the content).

ncartron, 2021-02-16:
WIP by @piggz - i've got a running version and this works super well - got detailed notifications sent from Amazfish (SFOS) to the PineTime

muppeth, 2021-02-17:
@ncartron Great to hear. Which version do you have in mind? Amazfish or Infinitime?

ncartron, 2021-02-17:
@muppeth running the latest Infinitime (0.13.0), along with a pre-release of Amazfish

petterhs, 2021-04-20:
Am I correct that this could be closed now that #257 is merged?

ncartron, 2021-04-20:
I believe so, at least for me i don't encounter the issue anymore for a while

willlovesbearz, 2021-04-22:
Yes, this should be closed.

closed 2021-04-22 by willlovesbearz