InfiniTimeOrg/InfiniTime #203: Moving the doc to InfineTime.io or to the Wiki Page

adrienluitot, 2021-02-10:
Hello!

I think the doc through md files like that is not really convenient.
It would certainly be more readable and easy to use if it was moved into the wiki system of Github and/or into the https://github.com/JF002/infinitime.io website.

JF002, 2021-02-16:
I understand it's not convenient for the end user. However, this is the easiest way for me to maintain the docs.

There are already PRs related to the improvement of the documentation, I still have to take a better look at them : #138 , #143 , #153 .  Feel free to comment them to give your opinion!

adrienluitot, 2021-03-01:
Hello!

I'm very sorry for such a late answer >.<

I think a doc like on #153 is good, it's very readable and it's still markdown files so the migration shouldn't be hard to do, and it should also be easy to maintain.

JF002, 2021-12-27:
The re-write of the documentation will be done in https://github.com/InfiniTimeOrg/InfiniTime/pull/838

closed 2021-12-27 by JF002