InfiniTimeOrg/InfiniTime #204: Notification popup does not show when an app is open, even if the screen is off

willlovesbearz, 2021-02-11:
I understand not having the notification popup when an app is in use and believe that to be a useful feature, but it stops being a feature once the screen is off.

JF002, 2021-02-16:
What do you mean? The notification should be displayed even if an app was open and the screen went off (sleep mode)?

ncartron, 2021-02-16:
Agreed with @willlovesbearz - just tested on 0.13.0 / Amazfish (SFOS):
- open an app
- force a notification to be sent by the phone
=> the PineTime vibrates, but no notification appears on the screen

same behavior when screen is off:
- open an app
- wait for the screen to go off
- force a notification to be sent by the phone
=> the screen wakes up, PineTime vibrates, but only the previously loaded app is displayed, not the notification

petterhs, 2021-02-16:
Yes it's currently just showing notifications when in the default Clock screen, even with screen off. We could add a acceptNotification boolean to each screen, so each app can "decide" if it wants notifications or not. Assuming we don't want notifications while playing a game for instance.

petterhs, 2021-04-20:
This issue is not present anymore and could be closed @willlovesbearz

closed 2021-04-22 by willlovesbearz