InfiniTimeOrg/InfiniTime #208: Can weather support be added?

hrmckay, 2021-02-14:
Can retrieving weather data through Bluetooth be added to the Pinetime?  This feature was added to the Pebble watch using Gadgetbridge.  A helper app on the phone called "Weather Notification" can be linked to Gadgetbridge and supplies the weather.  More information can be found here: https://github.com/Freeyourgadget/Gadgetbridge/wiki/Weather

Hugh

vbelloir, 2021-02-14:
Nice idea!

JF002, 2021-02-16:
Thanks for you idea! The integration of this functionality will be easier as Gadgetbridge already implements it. We just need someone to work on it :)

pfeerick, 2021-05-12:
joaquim's pinetimeLite v0.5.0 shows some experimental weather app capability, so progress is being made on this front ;) https://github.com/joaquimorg/PinetimeLite/releases/tag/liteV0.5.0

hrmckay, 2021-05-12:
Thanks for pointing that out.  However, it looks like they are using a custom version of Gadgetbridge.  The standard version of Gadgetbridge can supply weather by using a helper app.

Avamander, 2021-05-12:
> The standard version of Gadgetbridge can supply weather by using a helper app.

Also if the Android flavour has a weather provider.

bhibb, 2021-09-12:
Would love to see this added!

gnome71, 2021-11-27:
I started working on a weather app on the pinetime to get into writing displayapps. If you want to take a look:
[branch weather](https://gitlab.com/rfc_2324/InfiniTime/-/tree/weather)
It can be updated via bluetooth by a python app which is at:
[branch alert](https://gitlab.com/rfc_2324/wttr-py/-/tree/alert)
I use this script on my PinePhone to update it.
But! Its by far not finished. Work in progress ;-)

closed 2021-12-09 by JF002