InfiniTimeOrg/InfiniTime #210: Vibration on notification a bit too short/weak

muppeth, 2021-02-17:
Thanks a lot for vibration in notifications 😻 This makes pinetime much more of a daily driver for me. I did notice however that the vibration is very short and weak which makes me miss often the fact I got notification. 

In the future would be nice to have this configurable form within the watch but perhaps it could be raised a notch now?

grayrattus, 2021-02-20:
I'm not sure if it's possible to make it stronger but it's possible to tweak how long it lasts.
https://github.com/JF002/Pinetime/blob/develop/src/components/motor/MotorController.cpp#L20

Maybe it would be good to integrate it with the same widget as it is used for brightness.

JF002, 2021-02-21:
Yes, the length of the vibration can be configurable in the future :) 
For now, the motor is just activated for a specific amount of time (35ms, I think). 

As @grayrattus said, it would be fairly easy to build a configuration like the one for the backlight.

I plan on experimenting with the motor : drive the motor using a PWM to make the vibration smoother, and maybe implement multiple patterns.

ildar, 2021-03-14:
@JF002 , you surely can do. But why the duration is uint8_t? Then maximum vibration time is 255 ms?

zacharycarter, 2021-09-29:
This would really help me out being able to tweak the duration in userland without having to build infinittime or upgrade the firmware for every tweak. I have bad ADHD and bought the pine time with the intent of using it to keep track of time and my routine but I'm finding I miss the vibrations as well...

de0u, 2023-07-20:
Personally, for me the motivation behind purchasing a PineTime was that I hoped to work around the fairly feeble vibrating alerts from the Android phone in my pocket by having vigorous vibrating alerts on my wrist.  I am running the current firmware (1.13), and the text-message alert is sufficiently subtle that I may go back to not wearing a watch.

Reading around, there are other issues filed about vibration length (for notifications, alarms, and timers) that have been closed as duplicates of this one, so it seems quite a few people are interested in this.

sebpearce, 2023-12-12:
One idea that would be helpful (and maybe an easier step along the way to having full support for notification customisation?) is to allow choosing between the current single buzz and the wake alarm, i.e. where it keeps buzzing until you manually turn it off. This is how most watch timers work in my experience.

msugakov, 2025-04-07:
It's 2025 and on firmware 1.15.0 the vibration repetition still does not seem to be configurable. Is this correct?
There's https://github.com/InfiniTimeOrg/InfiniTime/pull/2235 in progress which seems to aim making the duration of one motor activation configurable. The configuration of number of repetitions (or the overall duration of such), or buzzing until turned off seems missing.

msugakov, 2025-04-11:
An attempt to improve this myself will require buying DevKit which with shipping and all taxes will cost me $50. It's also out of stock at the moment. Not to mention the skills required to make this happen.
I enjoy the product and wish the vibration for notifications will be more configurable. I offer to donate these $50 to whoever can make this improvement. Perhaps others who desire this feature would like to support too.

de0u, 2025-04-11:
It's not clear that InfiniTime is set up to accept GitHub donations, which I currently use to support two projects.

As I had suspected when I commented 2.5 years ago, without robust vibrations I stopped wearing my PineTime.  It's in its box somewhere.

Nitrowing, 2025-04-11:
> As I had suspected when I commented 2.5 years ago, without robust vibrations I stopped wearing my PineTime. It's in its box somewhere.

I mounted mine in a little enclosure, makes a nice little table clock.