InfiniTimeOrg/InfiniTime #212: Umlauts in notifications

stephanlachnit, 2021-02-21:
Currently it Umlauts (ä, ö, ü) don't seem to work the notifications. I guess this is true for not just Umlauts, but other letters as well (ß, å, etc). Would be good esp for European users.

JF002, 2021-02-24:
For now, only the character a-z, A-Z and 1-9 are included in the font. Other characters like é, è, ê, ä, ö, ü,... are not included, but can be included easily, which will make notification easier to read :)

moriel5, 2021-10-02:
As I had pointed out in issue #461, in my country we use an RTL language, and unfortunately, we use a completely different script, with many not understanding English, so we need lightweight fonts, in addition to possible solutions (like loading from external storage).

tootbrute, 2022-01-10:
We need more language support. I just tried using Pinetime in Taiwan. It has no support for Chinese characters.

If the message is English the message comes through. If the message is all Chinese with no English the notification isn't even sent to the watch.

mackay508, 2022-03-04:
[image]

Would be very useful to have more character support, quite often have to ignore notifications as the á è etc style accents are displayed.

Avamander, 2022-03-04:
I will rephrase here the reply I wrote under a PR that intended on extending the symbol range.

Including more and more characters using the current method is not sustainable. There are many languages that should be included but building them into the firmware consumes space much better spent elsewhere. PRs extending the symbol range without using the external filesystem not be merged in the current state.

I think it's very clear that this feature is in high demand and there's not much need for further advocacy, so leave a thumbs up instead. If someone does start implementing it and wants to discuss that, then that's of course fine.

ptman, 2022-03-07:
Support for larger loadable fonts would be great. But in the meantime, where could we apply translations? ä => ae, ö = >oe, å => aa, ü => ue, ...

ThatOneCalculator, 2022-04-14:
This happens with emojis, but in somewhat of a strange way. One emoji (🎶) gets rendered as `:notes:` which is fine, but 🤩😘 gets shown as `;-*`

[image]

[image]

aha999, 2022-05-17:
I also have this issue when reading messages. It would be nice if Transliteration was enabled by default or if the app detects your location and turns it on accordingly.
In Slovenia, we don't get letters in pinetime notifications like Š, Č, Ž, so UTF-8 support would be great.

moriel5, 2022-10-18:
Hmm... Now that we have support for resources on the external flash, could that be used for adding additional languages without flashing forks of the main firmware?

Boteium, 2022-11-17:
@tootbrute  I've just created a traditional Chinese fork. If you're still interested in PineTime, please visit 
https://github.com/Boteium/InfiniTime

tootbrute, 2022-11-18:
> @tootbrute I've just created a traditional Chinese fork. If you're still interested in PineTime, please visit https://github.com/Boteium/InfiniTime

omg yes. my wife can stop bothering me then. I assume that fonts will come to the main fork....eventually. I assume that European languages will be prioritized though.

I'll check out your fork. Thank you!

jmlich, 2023-10-18:
I can see that the international characters support is limited by amount of device memory. What about adding conditional compiling and build special image for language group. For example `-DENABLE_UKRAINIAN=1 -DENABLE_POLISH=1 -DENABLE_CHINESE` ?

tootbrute, 2023-12-18:
> @tootbrute I've just created a traditional Chinese fork. If you're still interested in PineTime, please visit https://github.com/Boteium/InfiniTime

can you update to the latest version of infinitime? i would be willing to pay you directly or donate to an open-source project like Infintime? it doesn't look like languages are coming very quickly to infintime.

Boteium, 2023-12-18:
@tootbrute I'll find time to do it after the new year eve.

I do have a latest personal build so it shouldn't be difficult.
The reason I didn't release it was that I've since make several breaking changes primarily to make more room, including removing even more apps and bma423 driver since my pinetime has bma425, and codepoint hack which would display nothing in Chinese unless a special version of gadget bridge is installed.
So, I do not think this version is appropriate for release but I'll make one for you in about two weeks. Btw, your unit also has bma425 i presume ?

Boteium, 2023-12-18:
Because this thread is still getting new activities, I would like to promote my pull requests.
I believe theses two are sufficient for supporting most languages.
https://github.com/InfiniTimeOrg/lvgl/pull/3
https://github.com/InfiniTimeOrg/InfiniTime/pull/1773
Please give it a thumb up or leave a comment if you're interested.

tootbrute, 2023-12-19:
> @tootbrute I'll find time to do it after the new year eve.
> your unit also has bma425 i presume ?

not sure. how to check?

i see they changed in 2021? https://www.ncartron.org/infinitime-blue-purple-elderberry---stability-amp-metronome-app.html

i think my pinetime is only about 1.5 years old, so should be the new one.

FintasticMan, 2023-12-19:
On the About page, in the Accel. field, it says what accelerometer you have.

tootbrute, 2023-12-19:
> On the About page, in the Accel. field, it says what accelerometer you have.

Thank you. Sadly mine just tells me ???

[image]

FintasticMan, 2023-12-20:
Unfortunately, that's likely to be caused by a hardware issue like the accelerometer being broken :-(... You can try running the battery down completely, sometimes that can help.

ThatOneCalculator, 2023-12-20:
> Unfortunately, that's likely to be caused by a hardware issue like the accelerometer being broken :-(... You can try running the battery down completely, sometimes that can help.

I have the same issue (happened after an update several months ago), and tried that, but it never ended up working again.

jmlich, 2024-11-26:
There is also transliteration in Amazfish. Right now, It needs to be turned on manually.

minacode, 2025-02-11:
With #2217 on the way (not merged yet), we can now get notifications from Apple devices. But there is no companion in between. The notifications are send directly from the OS. To support at least some characters, we would need an on-watch translation layer.
The same would be true for Apple media control, if it is done somewhere in the future.

mark9064, 2025-02-11:
I think it would be worth looking into diacritic support for latin languages, the amount of extra space needed might be relatively manageable and it provides support to many languages at once. For non-latin scripts though it's much more problematic space-wise, and loading a full system font into memory from the external flash is a no go as well. Loading into memory could work just for the notification screen though as there is a lot of free heap on this screen. Loading specific characters on demand could be possible but will probably lead to terrible performance when an entire message requires such characters.

Boteium, 2025-02-12:
> I think it would be worth looking into diacritic support for latin languages, the amount of extra space needed might be relatively manageable and it provides support to many languages at once. For [...]

yes, it's certainly possible to include common latin characters. that does not comsume much storage space anyway. however, our watch has very little amount of ram. so, it would be nice to let users choose what characters to include.

on demand loading is certain possible and I believe many other proprietary watches run on the nordic hardware are doing exactly this. as for our watches, we use lvgl. the official lvgl font engine load the entire font into ram and does not have on demand support. to implement this, one would need to write a custom lvgl font engine to fetch the characters on demand. it is possible but not easy.

i believe pr 1773 is the first step toward support multiple languages becuase this is the simplest implementation compare to on demand loading. it would be nice to have some insight from the maintainer so that we could improve this pr to meet this project's merge  standard.

minacode, 2025-02-12:
#1773

minacode, 2025-02-13:
The only other external text inputs are in the music app, right?

mark9064, 2025-02-17:
I think so? In that case there are just two locations to consider