InfiniTimeOrg/InfiniTime #214: Fitness tracking stats

stephanlachnit, 2021-02-22:
It would be cool if the watch could show activity stats when a fitness tracker is active, for example duration, distance, avg pace, etc.

I would like to contribute, and while I do have some C/C++ experience, I have no clue about BLE or anything InfiniTime related. I guess the first thing to do would be to know if there is something about this in the BLE spec.

stephanlachnit, 2021-02-23:
So from what I quickly gathered, this can be achieved with the RSCS BLE spec. A companion app would need to support sending the information via RSCS, and then we could collect it.

From the supported features, I think interesting is total distance and instantaneous speed are the only really interesting ones for us. Sadly I didn't find anything related to total duration or average speed. But I'm not sure, if this is actually the correct spec for this.

JF002, 2021-03-18:
Hi and thanks for your feature request!

By reading the description, I'm not sure if you want the PineTime to acquire fitness value and send them to a companion app (running on a smartphone) over BLE, or if you want the PineTime to connect to another device (activity tracker) and display data from this device on the activity tracker.

The good new is that both should be possible to implement.

If you want to send fitness values to a smartphone, you'll have to acquire them (using the HR sensor, motion sensor,... and send them to a companion app. You can check with Amazfish and Gadgetbridge, 2 open-source companion apps that support InfiniTime.

If you want the PineTime to connect to another BLE device, you'll need to work a bit more on the BLE side. For now, InfiniTime waits for a device to connect to it. In this case, you want InfiniTime to actually connect to another device. I know that NimBLE (the open-source BLE stack InfiniTime is based on) supports that mode, but there is no similar use case for now.
Then, you'll need to figure out the BLE API used by your external device. Even if there are many services that are standardized, many commercial devices implement a non-standard API, making the integration much more difficult.

stephanlachnit, 2021-03-22:
> By reading the description, I'm not sure if you want the PineTime to acquire fitness value and send them to a companion app (running on a smartphone) over BLE, or if you want the PineTime to connect to another device (activity tracker) and display data from this device on the activity tracker.

The later one. Basically the phone tracks your speed via GPS, and the watch shows you this information. I think the Bluetooth PAMS spec should provide this (I quickly skimed through the spec from https://www.bluetooth.com/specifications/specs/).

> If you want the PineTime to connect to another BLE device, you'll need to work a bit more on the BLE side. For now, InfiniTime waits for a device to connect to it. In this case, you want InfiniTime [...]

Hm, can't the fitness app connect to the watch? I'm not an expert, but I guess the initial setup would work similar to CTS and ANS.

limoncelli, 2021-05-23:
The FOSS Android App Fitotrack (on Fdroid) has quite a set of metrics it tracks. Via Gadgetbridge you're supposed to be able to connect to the heart rate sensor of infinitime via gadgetbridge. Maybe this would allow sending metrics the other way around as well. Fitotrack has intervals as well, which are interesting for running.

limoncelli, 2021-05-24:
This is the thread fitotrack thread https://codeberg.org/jannis/FitoTrack/issues/170 with some more information.

bearclaws8, 2021-08-13:
I posed this in item #383 but it may be applicable here too.

One possible phone app to pull from that I have used is an open source tracker, [OpenTracks](https://f-droid.org/en/packages/de.dennisguse.opentracks/) that is available on Fdroid. It captures much of the information you are looking for and can show it via notifications on the phone. I'm not sure if it has a way to pull the data from the app itself but if the PineTime could snoop the notification it may be able to display the data (and maybe stop the notification from showing up in the notification area of the watch, which this app inundates).

stephanlachnit, 2021-08-13:
>  I'm not sure if it has a way to pull the data from the app itself but if the PineTime could snoop the notification it may be able to display the data (and maybe stop the notification from showing up in the notification area of the watch, which this app inundates).

I don't think this a good idea, since this would need to be done individually for every app. It makes way more sense to implement a proper bluetooth interface, in particular the Physical Activity Monitor Service ([PAMS spec](https://www.bluetooth.com/de/specifications/specs/physical-activity-monitor-service-1-0/)) and the Physical Activity Monitor Profile ([PAMP spec](https://www.bluetooth.com/de/specifications/specs/physical-activity-monitor-profile-1-0/)).

Quoting from the PAMP spec:
> The Physical Activity Monitor is the device that reports physical-activity-related data to a Collector. The Collector is the device (for example, a mobile phone) that receives the physical-activity-related data from a Physical Activity Monitor. If supported by the Physical Activity Monitor, the Collector may also control the Physical Activity Monitor (for example, to start, stop, and delete sessions).
> [...]
> A Physical Activity Monitor instantiates the PAMS [1] and the Device Information Service [3].
> A Physical Activity Monitor may instantiate the User Data Service (UDS) [4].
> A Physical Activity Monitor may support the Device Time Server role (as defined in [6]) and instantiate the Device Time Service (as defined in [5]).
> A Collector shall support the Device Time Client role as defined in [6].
> [...]
> The Collector shall use the Physical Activity Monitor Service [1], the User Data Service [4], and the Device Time Service [5].
> The Collector should use the Device Information Service [3].

In this case, the fitness app is the Monitor and the PineTime is the Collector.

bearclaws8, 2021-08-13:
> I don't think this a good idea, since this would need to be done individually for every app. It makes way more sense to implement a proper bluetooth interface, in particular the Physical Activity [...]

@stephanlachnit I agree this is probably a terrible idea.  I have lots of those and silly ones as well. 😜

Asking out of ignorance, but seeking to understand, is the PAMP spec generally supported by many apps such that just by implementing that standard we could readily send/receive data?  Are there certain things we should look for in apps that would allow the PineTime to be a collector, or would a phone app need to enable this specifically?  I read through parts of the PAMP spec but the answer wasn't clear to me based on my very limited understanding.

stephanlachnit, 2021-08-13:
I'm certainly not am expert here, but I think the app needs to implement it specifically. And I'm not aware of any app supporting this configuration yet.

scandinave, 2022-01-19:
BLE spec seams to be a good idea. Il should be possible to make a android/ios app that can juste send this data via PAMS to the pinetime. 

But reading the spec, i see this : 

```
Use of this specification by anyone who is not a member of Bluetooth SIG is prohibited and is an infringement of the
intellectual property rights of Bluetooth SIG and its members.
```

This is not a problem?