InfiniTimeOrg/InfiniTime #215: Flashlight watchapp

Raupinger, 2021-02-23:
App that lights up the screen so the watch can be used as a (rather dim) flashlight. This sounds like a fun first issue. Maybe ill take a look at it myself once my exam period is over.

FlameKat53, 2021-02-23:
this in theory is easy to implement, sounds fun :)

FlameKat53, 2021-03-12:
here's the code for the Flashlight app. enjoy!
[Torch.zip](https://github.com/JF002/InfiniTime/files/6131368/Torch.zip)

Raupinger, 2021-03-20:
thanks. I got my hardware a few days ago. Haven't really found time to set up the development environment though

pfeerick, 2021-04-13:
There will be a flashlight app as part of 0.16... and is present in the develop branch now as of https://github.com/JF002/InfiniTime/commit/eb769fb60ecb8f96ecf6901082ec3f0610842af8.

zero-one01, 2021-05-02:
Flashlight is available in the 1.0.0 release a80e782 in the settings menu (swipe from left to right). I think this issue can be closed.

ptman, 2021-05-21:
yes, please close

closed 2021-05-21 by Raupinger