InfiniTimeOrg/InfiniTime #220: [FR] bootloader graphics as DFU

ildar, 2021-03-02:
I assume this requires a short TODO:
1. Link FW with the 0x8000 linker script
2. After Graphics flashing FW finishes its job, it should rollback the previous FW

JF002, 2021-03-06:
Is this a bug report or a feature request?

The target `pinetime-graphics` is not compatible with the bootloader, as it's linked @ 0x00. I wrote this tool to allow people with a devkit built before InfiniTime to install the bootloader logo in the external flash. It is not intended to be used as a DFU file.

It could indeed be done, but I don't think it's worth the trouble as the next version of the bootloader will have the logo built in and won't need to read it from the external flash memory.

ildar, 2021-03-06:
if `pinetime-graphics` is soon to be gone, then, yes, we can just give up
on it.
And, yes, [FR] in the subject is "Feature Request"

JF002, 2021-03-06:
> And, yes, [FR] in the subject is "Feature Request"

Oooh I thought that meant "French" but... it didn't make any sense either :)

closed 2021-03-15 by ildar