InfiniTimeOrg/InfiniTime #224: Bluetooth / BLE connection watch face symbol unreliable.

gillham, 2021-03-10:
The symbol on the watch face that indicates a Bluetooth connection loses state or fails to update.
Reproduce:
Press the button on PineTime to wake it up.
Connect with nRF Connect 2.4.8 on iOS.
Note symbol appears.
Press disconnect / connect a few times, it should turn on and off.
With nRF Connect in the connected state (and symbol on the watch face), scroll the watch face to the first menu.
Touch the watch app icon to go back to the watch face
Test connectivity / symbol by disconnecting/connecting in nRF Connect, sometimes it is stuck.
If it still works, while connected scroll watch face and go to the "list" icon (information app)
Use the button to return to the first menu
Touch the watch app icon to go back to the watch face
Test connectivity / symbol by disconnecting/connecting in nRF Connect, this is always stuck in my testing.

Going to different apps and back to the watch face will generally reset this and the symbol state will start tracking again.

Avamander, 2021-03-10:
What InfiniTime version?

gillham, 2021-03-10:
At least v0.14.1 (currently the latest).

PR #796 (2021-12-09): Connect and bond with a passkey
Add Feature - Connect and bond with a passkey

- [X] Passkey pairing - passkey is displayed on watch and entered on phone
- [X] Swipe down to clear passkey screen
- [X] Connection encryption
- [X] Connection bonding
- [X] Automatic reconnects to bonded peripheral\*
- [X] Trusted device on Android
- [X] Persist bond between reboots

\* ~~**Rebooting the watch will cause reconnect failures. You must delete the bond from the phone to reconnect/pair.** Persisting the bond between reboots will resolve this.~~

Closes #224 
Closes #302 
Closes #502 

There are several [... cut]
closed 2021-12-09 by JF002