InfiniTimeOrg/InfiniTime #225: Error in overflow calculation

Panky-codes, 2021-03-11:
```cpp
void DateTime::UpdateTime(uint32_t systickCounter) {
  // Handle systick counter overflow
  uint32_t systickDelta = 0;
  if(systickCounter < previousSystickCounter) {
    systickDelta = 0xffffff - previousSystickCounter; // Error? 
    systickDelta += systickCounter + 1;
  } else {
    systickDelta = systickCounter - previousSystickCounter;
  }
  ``` 
  In case of a timer overflow, we do an extra calculation to avoid an error. 
  
  But the offset used is `0xffffff` which is 2^24. But I would expect it to be `0xffffffff` which corresponds to 2^32 for `uint32_t`. 
  I can create a PR if you think you agree with the bug. Or let me know if I am missing something here.

JF002, 2021-03-18:
Thanks for your report!
However, I think the current code is right : the value `systickCounter` comes from `nrf_rtc_counter_get(portNRF_RTC_REG);` and `portNRF_RTC_REG` is `NRF_RTC1`. See https://github.com/JF002/InfiniTime/blob/develop/src/systemtask/SystemTask.cpp#L219.

According to the reference manual of the NRF52832, the RTC device is based on a 24-bit counter register 
[image]

What do you think?

Panky-codes, 2021-03-18:
You are absolutely right! I got confused with the type used. 
Now I have another question: 
In my PR #144 , I used the Freertos function `xTaskGetTickCount()` function to get the ticks, and ofc I also implemented an overflow check.  But I used the 32 bits overflow over there. After you mentioned this, I went and checked the implementation of the Freertos `xTaskGetTickCount` function in [port_cmsis_systick.c](https://github.com/JF002/InfiniTime/blob/ada942535718d48eec37cca4f50d678e7201dc67/src/FreeRTOS/port_cmsis_systick.c#L114). Over there they use that function as an interrupt handler for RTC and increment the counter. So if I use  `xTaskGetTickCount`, then 32 bits overflow will be applicable unlike using the `nrf_rtc_counter_get` right? 

Let me know your thoughts.

JF002, 2021-03-20:
I think you are right : the tick count returned by `xTaskGetTickCount()` is managed by FreeRTOS and, if I understand correctly, they use the whole 32 bits from the variable:
[image]

closed 2021-03-21 by Panky-codes