InfiniTimeOrg/InfiniTime #228: Tap to wake?

FlameKat53, 2021-03-12:
Hello, i would greatly enjoy the ability to tap to wake up my watch. here is the code and what you need to change it to.

code:
```
void SystemTask::OnTouchEvent() {
  if(isGoingToSleep) return ;
  NRF_LOG_INFO("[systemtask] Touch event");
  if(!isSleeping) {
    PushMessage(Messages::OnTouchEvent);
    displayApp->PushMessage(Pinetime::Applications::DisplayApp::Messages::TouchEvent);
  }
}
```
change to:
```void SystemTask::OnTouchEvent() {
  if(isGoingToSleep) return ;
  NRF_LOG_INFO("[systemtask] Touch event");
  if(!isSleeping) {
    PushMessage(Messages::OnTouchEvent);
    displayApp->PushMessage(Pinetime::Applications::DisplayApp::Messages::TouchEvent);
  } 
  else {
      if(!isWakingUp) {
      NRF_LOG_INFO("[systemtask] Screen tapped, waking up");
      GoToRunning();
      }
    }
}```

pfeerick, 2021-04-13:
joaq's recent PRs have incorporated this - both single and double tap wake up options, and should be in 0.16. It is present in the current develop code (https://github.com/JF002/InfiniTime/commit/eb769fb60ecb8f96ecf6901082ec3f0610842af8).

JF002, 2021-04-23:
This feature (single tap, double tap and wrist raise to wake) were added in 1.0 ;)

closed 2021-04-23 by JF002