InfiniTimeOrg/InfiniTime #242: Make fails due to FLASH overflow

FolsomMike, 2021-03-20:
Linking CXX executable pinetime-app-0.14.0.out
/home/ubuntu/pinetimedev/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: pinetime-app-0.14.0.out section `.text' will not fit in region `FLASH'
/home/ubuntu/pinetimedev/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
/home/ubuntu/pinetimedev/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 45680 bytes

JF002, 2021-03-20:
That's probably because you are building in DEBUG mode. In this mode, the binary size has become too big to fit in the internal flash memory. This is a side effect of adding new features :)

As a workaround, you can disable the debug mode by removing `-DCMAKE_BUILD_TYPE=Debug` or modifying it to `-DCMAKE_BUILD_TYPE=Release` in your CMake command line.

If you want to actually debug your build, you'll have to disable some parts of the code to reduce the binary side. For example, the icons of the navigation app take a lot of space : comment the content of the array `m_iconMap` in https://github.com/JF002/InfiniTime/blob/develop/src/displayapp/screens/Navigation.h#L148.

FolsomMike, 2021-03-20:
Thanks...that worked.

closed 2021-03-20 by FolsomMike

pur-jus, 2021-04-28:
I tried to comment the content of the array ``m_iconMap`` but the built result indicates a FLASH overflow of 148260 bytes  :grimacing:

```bash
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: pinetime-app-1.0.0.out section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 148260 bytes
collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/pinetime-app.dir/build.make:1407: src/pinetime-app-1.0.0.out] Error 1
make[3]: Leaving directory '/home/asuju/Work/InfiniTime/build'
make[2]: *** [CMakeFiles/Makefile2:336: src/CMakeFiles/pinetime-app.dir/all] Error 2
make[2]: Leaving directory '/home/asuju/Work/InfiniTime/build'
make[1]: *** [CMakeFiles/Makefile2:343: src/CMakeFiles/pinetime-app.dir/rule] Error 2
make[1]: Leaving directory '/home/asuju/Work/InfiniTime/build'
make: *** [Makefile:222: pinetime-app] Error 2
```

Where is my mistake?

In advance, thank you.