InfiniTimeOrg/InfiniTime #243: Smart Home watchapp

Raupinger, 2021-03-20:
This is probably something to work on down the line, but an App to control smart lighting via openhab would be nice to have. This would obviously need support in the companion apps which doesn't currently exist if I'm not mistaken.

JF002, 2021-03-20:
Smart home application was one of my original project! But building a whole application firmware deverted me from this goal :)

As PineTime can only communicate on BLE, you'll indeed need a companion app or a bridge device to bridge your pinetime to your smart home application (OpenHAB or Home-assistant in my case).

pfeerick, 2021-03-22:
Sounds like a perfect job for the pinecone (bridge device)! ;)

sporkbyspork, 2021-04-22:
For the glorious Pebble, there are companion apps like [PebbleTasker](https://play.google.com/store/apps/details?id=com.kodek.pebbletasker&hl=en&gl=US) or
[AutoApp for Pebble](https://play.google.com/store/apps/details?id=com.joaomgcd.autopebble&hl=en&gl=US).
You create the situation-aware menus on your phone and the watch shows them. Then you can use the almighty Tasker to react.

I dread the moment when my Pebble dies and there are no more replacements. So a comparable functionality would be a great way to get people to jump onto the PineTime bandwagon.

funnyflowerpot, 2022-01-12:
I would also be very much in favour of possibilities to add smart home capabilities to the PineTime!

This might be a bit of a tangent, but I wanted to share experiences briefly. In [a comment](https://github.com/InfiniTimeOrg/InfiniTime/issues/562#issuecomment-907613478) in a related issue @unrelentingtech was linking to a [project on Github](https://github.com/shmuelzon/esp32-ble2mqtt/) that implements a BLE/GATT to WiFi/MQTT bridge on an ESP32. I found the project a while ago, tried it out, and connected the PineTime to it. 

Summary: I was able to send notifications to my watch via MQTT with for example the shell script below. It worked quite well and liked it a lot! 

```
PINETIME_MAC="11:22:33:44:55:66"
MQTT_TOPIC="$PINETIME_MAC/AlertNotificationService/NewAlert/Set"

printf '\x00\x01%s\x00%s' "$1" "$2" | \
  od -tu1 -An | xargs | tr ' ' ',' | \
  mosquitto_pub -t "$MQTT_TOPIC" -s
```

To me this was an eye-opener for so many new possibilities and I would really love to see development into this direction! Maybe I can get into InfiniTime development later this year.

Also, as a more general observation, I noted that a BLE-MQTT-bridge (or anything similar) can remove the need of using a smartphone or companion app, which is a great advantage in certain situations. This of course depends on personal use case. However, it could be seen as trading dependency on the phone for location-based functionality (if the ESP32/etc. is permanently installed). This would be an advantage for some people (including me, who would love to get rid of wearing the smartphone all the time in favour of the PineTime).

cpyarger, 2022-05-06:
It would be nice to see this as something that can run as an addon via home assistant Something like an extension of the Passive BLE monitor integration,

cpyarger, 2022-05-06:
I have submitted an issue with a new sensor request, in order to have the infinitime watch added as a supported sensor with the passive ble monitor integration. Maybe if there is enough chatter on the issue, it may be considered. [New Infinitime Sensor](https://github.com/custom-components/ble_monitor/issues/846)

JF002, 2022-05-06:
@cpyarger  I'm a user of Home Assistant, but I don't know about that passive BLE monitor Integration. If I understand correctly, this would allow to detect the PineTime and even poll for step or HR data and use those data in HA?

cpyarger, 2022-05-06:
> @cpyarger I'm a user of Home Assistant, but I don't know about that passive BLE monitor Integration. If I understand correctly, this would allow to detect the PineTime and even poll for step or HR data and use those data in HA?

Yes. That is the idea. Also setting the time. And acting as a endpoint for notifications

JF002, 2022-05-07:
That's really nice ! I hope someone will consider working on it, then!