InfiniTimeOrg/InfiniTime #247: Digital clock  - order of composition

pfeerick, 2021-03-22:
Noticed with 0.15 (although it exists as at least as far back as 0.12) that the digital clock watchface is first populated with placeholder text / icons, and then the final display set, when returning to it from the menu (not from sleep). It became more obvious with the 0.15 version as there is a flash of colour up the top as the charge and bluetooth icons appear and disappear (they disappear to reflect the true state of the device). 

[image]

petterhs, 2021-04-20:
This was fixed in #258 and can be closed now

pfeerick, 2021-04-21:
Thanks for the reminder... yes, this has been gone for a while now!

closed 2021-04-21 by pfeerick