InfiniTimeOrg/InfiniTime #248: Linker Errors

jones139, 2021-03-23:
On a fresh clone of the 'develop' branch:
> cmake -DCMAKE_BUILD_TYPE=Debug -DARM_NONE_EABI_TOOLCHAIN_PATH=/usr/local/nRF5/gcc-arm-none-eabi-9-2020-q2-update -DNRF5_SDK_PATH=/usr/local/nRF5/nRF5_SDK/nRF5_SDK_15.3.0   -DUSE_OPENOCD=1 -D NRF_LOG_ENABLED=0 ../
> make -j pinetime-app

Result:
> [100%] Linking CXX executable pinetime-app-0.15.0.out
> /usr/local/nRF5/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: pinetime-app-0.15.0.out section `.text' will not fit in region `FLASH'
> /usr/local/nRF5/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
> /usr/local/nRF5/gcc-arm-none-eabi-9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 84828 bytes
> collect2: error: ld returned 1 exit status
src/CMakeFiles/pinetime-app.dir/build.make:1168: recipe for target 'src/pinetime-app-0.15.0.out' failed
> make[3]: *** [src/pinetime-app-0.15.0.out] Error 1
> CMakeFiles/Makefile2:378: recipe for target 'src/CMakeFiles/pinetime-app.dir/all' failed
> make[2]: *** [src/CMakeFiles/pinetime-app.dir/all] Error 2
> CMakeFiles/Makefile2:385: recipe for target 'src/CMakeFiles/pinetime-app.dir/rule' failed
> make[1]: *** [src/CMakeFiles/pinetime-app.dir/rule] Error 2
> Makefile:252: recipe for target 'pinetime-app' failed
> make: *** [pinetime-app] Error 2

I am not sure what I have done wrong - this method used to work, and I think it is consistent with the build guidance in the documentation (sorry if I have missed something!)

Avamander, 2021-03-23:
Duplicate of https://github.com/JF002/InfiniTime/issues/242

jones139, 2021-03-23:
Yes, sorry!

closed 2021-03-23 by jones139