InfiniTimeOrg/InfiniTime #261: Notifications don't show all utf-8 characters

mareksamec, 2021-04-09:
I have noticed that characters like ščřžěšÅÄÖ are omitted from the notifications shown on PineTime. I also tested this in Gadgetbridge by sending messages with these characters. It does not seem to be Gadgetbridge issue but I cannot rule it out 100%.

Please let me know if there is any way I can help with debugging, testing etc. Sorry if this has already been fixed in the newer releases.

InfiniTime 0.14.0
Gadgetbridge 0.55.0 (patched version by Joaquim)

joaquimorg, 2021-04-09:
The Infinitime font don't have the complete characters set for international languages.

https://github.com/JF002/InfiniTime/tree/develop/src/displayapp/fonts#readme

JF002, 2021-04-10:
That's a know ... missing feature. ITF-8 defines 1,112,064 valid characters and adding all of them into the font would probably fill the memory of the pinetime several time! For now, we've only included basic ASCII characters and cyrillic ones. It's still missing all accentuated letters from French Language (éèêë), german characters (https://github.com/JF002/InfiniTime/issues/212) and many other characters from other languages.

As the flash memory of the device is nearly full right now, we'll have to find a smart way to implement internationalization into infinitime :)

Avamander, 2021-04-10:
> As the flash memory of the device is nearly full right now, we'll have to find a smart way to implement internationalization into infinitime :)

:smiley: fonts stored on LFS loaded by LVGL's storage driver

mareksamec, 2021-04-10:
> That's a known... missing feature. ITF-8 defines 1,112,064 valid characters and adding all of them into the font would probably fill the memory of the pinetime several time! For now, we've only included basic ASCII characters and cyrillic ones. It's still missing all accentuated letters from French Language (éèêë), German characters (#212) and many other characters from other languages.
> 
> As the flash memory of the device is nearly full right now, we'll have to find a smart way to implement internationalization into infinitime :)

Too bad :-D Not sure how difficult would it be, but perpahs the Gadgetbridge could send letters and strip the accents. For instance in Python I would use something like [unidecode](https://pypi.org/project/Unidecode/ )

ptman, 2021-05-21:
can a larger font be stored on the SPI flash? where could we find statistics on most used glyphs?

lafriks, 2021-06-23:
> can a larger font be stored on the SPI flash? where could we find statistics on most used glyphs?

I would say that's really region dependent, I wonder how much would it be to include at least European languages?

Riksu9000, 2021-07-02:
There are 64 letters in the Latin-1 Supplement unicode block. I presume that includes a lot of the characters that people want.
Adding these characters in range 0xC0-0xFF increases binary text size 1832.
Removing the two mathematical symbols from the range surprisingly doesn't change the binary size at all.
https://en.wikipedia.org/wiki/Latin-1_Supplement_%28Unicode_block%29

Bloodaxe95, 2021-07-25:
I've noticed this issue as well, specifically with the letters Æ,Ø and Å. It would be pretty awesome to have my native letters supported :-D But from the posts above mine it seems there isn't a super-simple solution to this issue :-(

esijg, 2021-09-06:
I am experiencing the same thing with Icelandic special characters. I can alleviate that a bit by enabling transliteration in gadgetbridge but that does not cover all the cases sadly. 

```
string:			éþýúíóöáæð
transliterated:	eyuiooaae
```
Gadgetbridge seems to transliterate all of them correctly except for **þ** and **ð**. Which should transliterate to **th** and **d**. Now I do realize that this is an impossible task for the watch to store all these fonts, but would it be possible to make infinitime handle transliteration instead of hoping for 3rd party apps to handle it correctly? or is that asking too much of it?

Avamander, 2021-09-06:
> but would it be possible to make infinitime handle transliteration instead of hoping for 3rd party apps to handle it correctly?

Nope, too language-specific. Open an issue on Gadgetbridge's repository about it being incorrect in your language.

esijg, 2021-09-06:
> Open an issue on Gadgetbridge's repository about it being incorrect in your language.

Done. here's the link for anyone interested. https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2388

kieranc, 2021-10-01:
Duplicate of #212 and known issue, can be closed?

esijg, 2021-10-01:
Leaving a note here for future readers. Fixing/improving transliteration in Gadgetbridge seems to be super straight forward and they are quick to fix it once pointed out. I opened an issue on their issue tracker and it was fixed in couple of days. It just arrived in f-droid couple of days ago even and works wonders.

closed 2021-10-02 by Avamander

tomechio, 2022-02-26:
Hi, 
I created and font pack with the excellent instructions included in Infinitime to support special characters of Finnish, Swedish and Polish languages.
You can download it from [here](https://github.com/tomechio/Infinitime_FI_SE_PL_support)