InfiniTimeOrg/InfiniTime #262: Version 0.16.0

joaquimorg, 2021-04-09:
I have notice some problems with the touch when is in Polling mode.
Testing InifiPaint and Pong they freeze making the watchdog go into action.

JF002, 2021-04-10:
I've also noticed some glitches around the handling of the TWI driver (probably sleep/wakeup) since the merge of your BigRewrite PR and my motion sensor branch.  I'll try to debug them this week-end :)

JF002, 2021-04-10:
For anyone wondering where the 0.16.0 version is : it's not released yet ! It's the version of the *develop* branch and, as you can see, it's not stable enough to be released yet ;)

joaquimorg, 2021-04-10:
I think we have to go back to the original TWI, I reverted the code and there are no more blocks in the apps.
bma421 starts well and does not block on restart.

joaquimorg, 2021-04-10:
I created a PR #263 with some corrections, for those I already tested I do not have blocks of the apps, the BMA is working.

closed 2021-04-16 by joaquimorg