InfiniTimeOrg/InfiniTime #267: Documentation should mention Python modules

Caton101, 2021-04-14:
I am new to the InfiniTime and decided I would help out. I encountered an issue when trying to build InfiniTime because the build documentation never mentioned required the `cbor` and `intelhex` modules for Python 3. I recommend adding all required modules to the list of [build dependencies](https://github.com/JF002/InfiniTime/blob/develop/doc/buildAndProgram.md#dependencies).

I have already started this change inside of PR #266. Currently, I only listed the `cbor` and `intelhex` modules. I would appreciate the help with documenting any other Python dependencies.

pfeerick, 2021-05-12:
Is this resolved now that #266 is merged?

tmilburn, 2021-08-19:
I also had to install the cryptography python module.

closed 2021-10-02 by Avamander