InfiniTimeOrg/InfiniTime #270: Proposal: Night Mode / Time dependant brightness

qbinary, 2021-04-16:
I would really like to see something like a night mode or a time dependant brightness value so that the PineTime dims down at night and goas back to full (/last set) brightness in the morning.

JF002, 2021-04-17:
The pinetime does not have a luminosity sensor, but increasing / decreasing the backlight at some pre-defined (or configurable) time should not be difficult to implement.

qbinary, 2021-04-19:
I really would like to implement this feature but I am currently very bussy with my abitur exams and do not have enought time to get warm with infinitime. I already have my devkit hooked up and successfully compiled and flashed infinitime but until my last exam I have to restrict my contributions to issues :(

KoalaV2, 2021-11-06:
Found this issue relevant so including my suggestion here. Making the display yellow / darker towards the night would also be nice since all my other devices have this and opening a bright blue / white display during the night to check the time is a bit annoying.

Eloitor, 2022-08-19:
I'm also interested in this feature.

Riksu9000, 2022-08-19:
For me the brightness setting is very situational and hardly time dependant. How would overriding the brightness be handled? Wouldn't it be annoying if it just changed on it's own after you specifically set it to some level?