InfiniTimeOrg/InfiniTime #272: Feature Request: Ability to clear notifications

GIJack, 2021-04-17:
I am requesting adding a feature to clear notifications after they are read. Perhaps a swipe to the right?

JF002, 2021-04-17:
That would be a nice improvement, indeed. As far as I know, the standard Alert Service protocol does not provide this feature. We'll probably have to define a new BLE API for that.

GIJack, 2021-04-17:
Can just a clear buffer command that clears them locally? I mean wasp os does this already.

But a BLE API to acknowledge alerts on the watch without having to remove the phone from your pocket would be super sweet.

Avamander, 2021-04-17:
> That would be a nice improvement, indeed. As far as I know, the standard Alert Service protocol does not provide this feature. We'll probably have to define a new BLE API for that.

This can theoretically be lifted from the Pebble notification protocol and its GB implementation to a large extent. Currently GB keeps track of what notifications on the phone are dismissed and dismisses them from the Pebble, it wouldn't be too difficult to create a two-way sync.

First step would be making it possible to assign an unique ID for each notification on the watch. Second step would be a way to both update and delete notifications by ID.

JF002, 2021-04-17:
> Can just a clear buffer command that clears them locally? I mean wasp os does this already.
Yes, indeed, we could begin with clearing the notifications locally !

To acknowledge notifications from the watch, @Avamander is right!

petterhs, 2021-04-20:
Related to #176 

Is the goal just to remove the notification only on the device or to also clear it from the phone?

pfeerick, 2021-04-21:
Since the comparison is made with wasp-os, I think it's just clear locally, which is done on wasp-os by pulling down the notification screen done one more time once all the notifications are shown, and it then asks if you want to clear them or not.

GIJack, 2021-04-21:
Both actually. My primary concern is being able to clear locally like wasp OS.

That said, I would like as a stretch goal to clear the notification on the phone. I mean, my goal for a smartwatch is just a remote control for a smartphone that also tells time.

Avamander, 2021-10-02:
I'll consider this a duplicate of https://github.com/InfiniTimeOrg/InfiniTime/issues/272. Syncing dismissals to companion phone should probably be a separate issue, if there's any need for that.

closed 2021-10-02 by Avamander

pfeerick, 2021-10-03:
It's a duplicate of itself?