InfiniTimeOrg/InfiniTime #273: Feature Request: Improved Notifications

GIJack, 2021-04-17:
I am asking for a few features for notifications
1. Notifications should have the following headers/properties App Name -> Title -> Contents. Such as Email -> New Email from Blag -> Email contents
2. Configuration of what gets displayed. App name, Titles, Contents

JF002, 2021-04-17:
The notification implementation in InfiniTime allows to send a notification **title** and a notification **message**. If the notification does not contain a title, a default title ("Notification") is displayed.

The content of the title and the message is set by the companion app. 
Amazfish already implement this title/message protocol:
[image]

Gadgetbridge only sends the message for now, but I talked with them, and they'll probably add the title in a future release.

GIJack, 2021-04-17:
I also need configuration of what is displayed. Per app, if possible.

As of now, it displays all messages, which can be a privacy concern. If possible I'd love to set it, so when I get signal/text messages, it just shows I got a text message and from who, sans content.

Avamander, 2021-04-17:
> I also need configuration of what is displayed. Per app, if possible.
> 
> As of now, it displays all messages, which can be a privacy concern. If possible I'd love to set it, so when I get signal/text messages, it just shows I got a text message and from who, sans content.

This is something more of a companion app feature request. Filtering on the watch side once it has already been sent over the air sounds like placebo. Is it not?

JF002, 2021-04-18:
I also think this should be done on the companion app side. I think Gadgetbridge already allows you to choose which notifications should be sent to the watch.
InfiniTime has no knowledge of the app from the phone that sent the notification, it just receives text messages to be displayed.

davorvr, 2021-04-21:
> I am asking for a few features for notifications
> 
> 1. Notifications should have the following headers/properties App Name -> Title -> Contents. Such as Email -> New Email from Blag -> Email contents
> 
> 2. Configuration of what gets displayed. App name, Titles, Contents

I have posted a feature request regarding this on the Gadgetbridge issue tracker:
https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2266

JF002, 2021-04-23:
Looks like this feature will be released in the next version of Gadgetbridge. I guess we can close this issue here.

closed 2021-04-23 by JF002