InfiniTimeOrg/InfiniTime #274: 1.0 reverts to previose release on reboot

DIWesser, 2021-04-17:
Pretty much what the title says. I can install 1.0 on my unit nRF Connect, but when I reboot the device it reverts to 0.15. Not sure if this is related to the validate app missing from this release?

pfeerick, 2021-04-17:
The validate app isn't missing... it's moved... swipe left to right to bring up the quick actions screen, press the gear cog (bottom right), swipe bottom to top to scroll to the second page, press firmware, and then validate it.

If you don't validate, and the watch resets for any reason, it is supposed to automatically revert to the previously installed firmware - recovery mechanism for bad firmware update or bugged firmware.

DIWesser, 2021-04-17:
Oops, so it is. Thanks.

closed 2021-04-17 by DIWesser