InfiniTimeOrg/InfiniTime #277: Feature Request: Allow waking with a double tap, even when Wake Up is set to "Raise Wrist"

panos, 2021-04-17:
I get that it doesn't make perfect sense, but the accelerometer is not infallible. That can make using InfiniTime much harder.

I think that it'd make sense to enable waking by double tapping, even in this mode, instead of only allowing people to choose either "Raise Wrist" or "Double Tap". After all, it's considerably harder to accidentally wake the watch in comparison to "Single Tap" and many people familiar with smartwatches are already conditioned to that feature by default, so I'm not sure why the user shouldn't get to enjoy the best of both worlds.

panos, 2021-04-17:
(If that makes sense and this *is* something that should get implemented, may I give it a shot and do it myself?)

JF002, 2021-04-17:
Yes, I think that makes sense! Wake up 'sources' are mutually exclusive for now, indeed. We can either add a 'double tap + wrist' or add the possibility to select multiple wake up source.
You can of course implement it yourself and create a PR about it :)

PR #290 (2021-07-14): Multiple wakeup sources
This PR should add the ability to set multiple wakeup sources.

Warning: I do not have a devkit, only a sealed one. I am not even sure that this compiles, so beware.
Also, does anyone know if there's a better way to know whether a checkbox is checked?

This would (try to) fix #277
closed 2021-07-14 by JF002