InfiniTimeOrg/InfiniTime #282: Feature Request: Call Log App

GIJack, 2021-04-17:
I request an app for Calls/Logs
* display ongoing call with ability to hangup/hold/mute
* Log of previous calls and type, incomming,outgoing,missed.
* Ability to clear the list
* Ability to call back missed calls

pfeerick, 2021-05-12:
A lot of this is going to depend on a companion app to support it... i.e. ability to initiate a call, ability to mute or hangup existing call. A basic five/ten entry call log would be interesting... but at the same time I really wonder if it's warranted since that is just a duplication of what your phone is doing already... Having said that, if you have a headset/wireless earbuds/handfree speaker I can see the benefit since it allows a remote interface to the phone...

Avamander, 2021-05-12:
Android has locked down access to call history for a while now, it is very sensitive. Same goes for all active call or ringer control. I'm sure there are things that can't actually be done on newer Androids in terms of this functionality...

That aside, it would be a nice app, but functionality should stand behind feature flags that the companion app can toggle. Otherwise it's going to be terrible UX.

I suspect the situation is even worse on iOS.

pfeerick, 2021-05-12:
Since the call history could be based from the call events received by the watch, that isn't such an issue. I see the bigger issue being that most of this functionality isn't exposed over GATT - proper actual call related control seems to be in the domain of HSP (Headset Profile), thus custom services and companion support needed to recreate this functionality.

GIJack, 2021-05-13:
>if you have a headset/wireless earbuds/handfree speaker I can see the benefit 

Yes. Between motorcycles, bicycles, electricskateboards with BT helmets, wireless headphones, the watch becomes a companion to a headset/BT helmet to act as a controller for when one is in motion.

This is actually what I use a smart watch for. So this is my primary, and perhaps only concern.