InfiniTimeOrg/InfiniTime #283: Switch to battery app when charging

Kozova1, 2021-04-17:
The Pinetime Lite fork has this feature where it switches to the battery app when charging.  
This could be a very nice feature.

pfeerick, 2021-04-19:
Yes please! I was going to suggest and loot at writing a simple vibrate when put on charge or taken off charge, i.e. like the pebble, but I think this is a much better idea!

qbinary, 2021-04-21:
I would propose to just show a popup like when a notification is received. But to make that look good the line should fill faster. I made this #306 issue to address that.

moriel5, 2021-07-20:
I personally prefer to see the battery percentage at all times on the home screen, in a similar fashion to how LineageOS shows it

Riksu9000, 2023-01-30:
Would this address some problem, or what's the purpose of this?