InfiniTimeOrg/InfiniTime #287: Heart rate measurement start/stop and sleep

JF002, 2021-04-18:
When the HR sensor is measuring and the watch goes to sleep, on wake up, the sensor is still running, but the HR app shows that the measurements are stopped.

How to reproduce:
 - Go into the HR app and start measurement. The watch won't go to sleep as long as it's running
 - Push the button to go back to the watchface and put the watch to sleep
 - The watch goes to sleep, the HR sensor is stopped
 - Push the button to wake up the watch, the HR sensor is started again
 - Go into the HR app and notice that the measurements are stopped. 

This is not good : the hr sensor is running, but the HR task does not process the measurements.

Now, I asking what should be done here:
 - Continue the HR measurement in the background, even when the watch is sleeping? There is very little point to do this now, as the measurements are not stored/send to the companion app.
 - Automatically restart the sensor AND the data processing on wake up
 - Do not restart the sensor and do not automatically process the data on wake up

Any opinion on this?

vbelloir, 2021-04-18:
Like for the step counter and wrist motion sensor, HR sensor should continue measurement in background. 
It takes some times to get a value, and if user start HR app, I think this is because he wants a value on clock app when he rotate his wrist, or push the button.

pfeerick, 2021-04-18:
Do you mean it doesn't measure, or just that it's starting from scratch again... as I just ran through those steps... and after the initial sampling, it showed the HR again... so for me at least... I can't reproduce that exact sequence... 

Since there is no sending to a companion app at present, I don't see the point in continuing measurement either. Except that the user hasn't stopped it. Adding another button 'continuous' would make sense then... (as well as no. 3 if it's not continuous) but still... for what reason (other than flattening the battery quicker atm)? 

But until HR is refined more (i.e. periodic capture, storage, sending to companion app - wait I thought amazfish at least read the HR?) I think this is a sticky problem... doing no. 2 would at least make it so the hr monitoring stays active whilst the watch is on until the user turns it off... i.e. make it consistent with the last user instruction - although then you also need the display on long enough to see it if using the watchface that shows HR...

vbelloir, 2021-04-18:
AS @pfeerick says, in fact, I can't reproduce the issue.
With the watch on my wrist, I started HR app, and go back to the clock app. I got, few second later a HR value. 
I put 5minutes the wtatch on a table. HR sensors was stopped (no green light).
After that delay, I put back the watch on my wrist, and wake up it pushing the button. Few seconds later, I got an HR value.  
And HR app says that it is started.  
For me, the behavior is good. Maybe we could let the sensors on even if the watch is sleeping. I don't know what other smartwatch does.

JF002, 2021-04-18:
You are right, I got confused by the fact that the color of the measurement is green when the measurement is started, but it's grey when you leave the app and run it again. That's probably a small display bug. 
But measurement is consistent with the display, ok!

There is another bug : when you start the measurements, the watch won't automatically go to sleep. But when you leave the app and the watch go to sleep (manually or not), wake up and run the HR app again, the watch will automatically go to sleep after the defined timeout value.

JF002, 2021-04-18:
Fixed in https://github.com/JF002/InfiniTime/commit/a7114f21b7d1f8ae29e0e005757b42882917ac3d. I think I can close this issue.

closed 2021-04-18 by JF002