InfiniTimeOrg/InfiniTime #295: Visual glitch when turning on the screen after initiating a firmware update

panos, 2021-04-19:
## Steps to reproduce

- Open an app (e.g. Music)
- Wait for the screen timeout.
- Initiate an update via Gadgetbridge.
- Open the screen.
- The UI bugs out. It'll bug out even more if you use gestures.

joaquimorg, 2021-04-19:
Please indicate the version you are running, if possible an image of the problem.

joaquimorg, 2021-04-19:
I think that RC2 solves the reported problem.

panos, 2021-04-19:
> I think that RC2 solves the reported problem.

I managed to reproduce it immediately after updating to RC2 too. I'll give it another shot as I'm setting up my development environment.

petterhs, 2021-04-19:
Fix should be in RC3

panos, 2021-04-19:
I completely missed this.

panos, 2021-04-19:
I'll give it another shot and try to break it then promptly close the issue.

joaquimorg, 2021-04-19:
> I think that RC2 solves the reported problem.

Yes the fix is in RC3, my mistake.

JF002, 2021-04-20:
@panos Do you confirm this issue is fixed in RC3?

panos, 2021-04-21:
Nope. Repeating the steps I brought up after flashing and validating RC3 seems to cause the same issue.

[image]

panos, 2021-04-21:
Sort of. I scrolled away from the "Firmware update" screen, turned off the screen and turned it back on. That resulted in a crash.

panos, 2021-04-21:
It's not the exact same as the thing I reported initially, which has been resolved, but it's just a small difference in the steps.

pfeerick, 2021-04-21:
I can reproduce it... one nice thing with DFU being able to recover mid update now though is that after the reset the update will auto restart (with nrf connect, at least)... gadgetbridge is refusing to do updates... unknown element or some other obscure and unhelp error message.

panos, 2021-04-21:
> Gadgetbridge is refusing to do updates... unknown element or some other obscure and unhelp error message.

Disconnecting and reconnecting the watch helps.

pfeerick, 2021-04-21:
lol... I have no luck ever getting gadgetbridge to do an update... only had it even attempt to do it once... and that failed... I try it every now and then when doing different DFU updates and it spits that error out... so I stick with NRFConnect as it consistently 'just works'.

panos, 2021-04-21:
> lol... I have no luck ever getting gadgetbridge to do an update... only had it even attempt to do it once... and that failed... I try it every now and then when doing different DFU updates and it spits that error out... so I stick with NRFConnect as it consistently 'just works'. I'll certainly give that a try next time, as I do want to give it a chance ;)

It's a bit bootleg, but you can also try renaming the `.bin` file in the releases and appending a `.fw` prefix. That's what I did when I needed to get out of Pinetime-Lite and use Infinitime instead.

JF002, 2021-04-23:
I'll close this issue. Please open another one regarding the crash during the OTA when leaving the firmware update screen if it's still present in 1.0.

closed 2021-04-23 by JF002