InfiniTimeOrg/InfiniTime #296: Stopwatch lap counter bug

pfeerick, 2021-04-19:
Lap counter now starts at 21, increments properly to 29, then goes to 210. 

I suspect there is a '2' literal where it doesn't belong. 

1.0.0 RC3

JF002, 2021-04-19:
Thanks @pfeerick ! Fixed in https://github.com/JF002/InfiniTime/commit/410cc821f46acfaa34e66a801301c708a5f12625.

closed 2021-04-19 by JF002

Peetz0r, 2021-04-20:
...it's even on the photo I took :')

Oops. I meant to have `#%2d` instead of `#2%d` there.

I feel blind now. Thanks for spotting and fixing.

pfeerick, 2021-04-20:
He he... Dang it... HIDE THE EVIDENCE!!! I'm just sad I saw this _after_ I went to bed, so wrote the issue and was looking at the code on the phone... otherwise I would have submitted a PR to fix it! :-P `#%2d` probably would have been better... 9 => 10 would shift alignment... but it isn't that big a deal.

JF002, 2021-04-20:
Oh right, I just change to `#%d`. I'll add the `2` and release RC4 :)