InfiniTimeOrg/InfiniTime #297: Go back to Clock screen after sleep

qbinary, 2021-04-19:
I think it is really annoying that infinitime stays in menu screens after waking up from sleep. I would propose to go back to the Clock app after sleep. The only exceptions should be the "real" apps like stopwatch or music.

petterhs, 2021-04-19:
Agree, this would be nice

nanch, 2021-04-20:
It's a reasonable idea, whether "go to watch face after screen dim" is enabled could be configured in user settings with a reasonable default.

P.S. IMO, you should prefix feature requests with "Feature Request: ".

qbinary, 2021-04-20:
> P.S. IMO, you should prefix feature requests with "Feature Request: ".

I thought this is done through tags!? But if that is common sense i will do it in the future.

pfeerick, 2021-04-21:
> I thought this is done through tags!? But if that is common sense i will do that in the future.

It is, but I think at the moment we're trying to make it easy to tell at a glance if an issue is a bug that could be blocking 1.0.0 from going to release, or something else that can wait.

Bugsbane, 2021-11-05:
This is a huge one for me. I mainly use my PineTime as a watch (surprise, surprise), but 90% of the time when I go to look at it it's not on the main time screen (due to accidental screen taps throughout the day, using other apps etc) On some apps you can still see the time, albeit tiny, in one corner, but often it'll be on some app that doesn't even show the time and instead of being a quick glance I end up having to use my other hand to repeatedly push the physical button before I can see the actual time on my watch...

Riksu9000, 2022-03-24:
Usually I would close the newer one of duplicate issues, however since #570 has had more discussion, I think it makes more sense to continue there.

closed 2022-03-24 by Riksu9000