InfiniTimeOrg/InfiniTime #299: Update to latest version of mynewt-nimBLE

MysteriousLog6, 2021-04-20:
There will be a few errors during build time , Which can be easily fixed.
Will be doing a pull request for the Build Time Fixes once the mynew-nimBLE is updated.
(I dont know yet how to do selective updates.

JF002, 2021-04-20:
Yes, upgrading to the last stable version of nimble would be a good idea!

For now, the whole repo of Nimble is copied into the project. But I would like to take the opportunity to use *git submodules* for nimble, the same way we did for LVGL.

MysteriousLog6, 2021-04-21:
I did try updating the entire stack (Please do correct me , I am a bit new to embedded :) I was having a lot of build time issues , we might have to modify the stack and not just keep the new version , I will try to redo updating it and fix the Build errors.

MysteriousLog6, 2021-04-21:
nvm it now works maybe Linux was just not happy yesterday?

MysteriousLog6, 2021-04-21:
Did exactly:
`make clean
make -j4 pinetime-mcuboot-app
`
And it worked!!
Maybe linux wasn't happy yesterday.

geekbozu, 2021-10-03:
This is on the todo list, I need to finish amassing information about what our settings actually are and get nimble to respect them outside of its folder then this will be relatively trivial...once that not trivial thing is done! 
With the upstream BLE fixes that we merged with 1.6.0 it will be nice to follow upstream again as well.