InfiniTimeOrg/InfiniTime #300: Watch not turning on after updating to 1.0.0-RC2

BeneCollyridam, 2021-04-20:
I have two (sealed) watches and the first one I flashed the update and everything works well. I then flashed it to my second watch and now it does not turn on. I of course know this could happen, and I accept the responsibility, I just thought I'd share my experience.

Avamander, 2021-04-20:
Old bootloader?

Caton101, 2021-04-20:
I've had similar issues when updating over Bluetooth where the connection would drop while flashing. It sounds like you are using the old bootloader since the new one can recover from a failed update.

BeneCollyridam, 2021-04-20:
I haven't updated the bootloader since I got the watch, so that is probably it.

petterhs, 2021-04-20:
> I haven't updated the bootloader since I got the watch, so that is probably it.
> 
> Is there a possibility to recover it or is it permanently bricked?

One way is to let the battery drain, then put it in the charger. Or you can open it  and either power cycle it or flash with SWD

closed 2021-04-20 by BeneCollyridam