InfiniTimeOrg/InfiniTime #302: Watch does not automatically reconnect to GadgetBridge after turning airplane mode on and off

GithubUser5462, 2021-04-20:
InfiniTime version: 1.0.0 RC3
Gadgetbridge version: 0.56.1

Steps to reproduce:

1. Pair InfiniTime with Gadgetbridge, enable the "Reconnect automatically" setting in GB
2. While the two are connected turn  airplane mode on and then off after the watch is disconnected
3. GB tries to reconnect, first with status "Connecting" and then gets stuck with "Waiting for reconnect"

The workaround I have found is to fully disconnect in GB - status "Not connected" and then pair the watch again.
Restarting the watch or Gadgetbridge did not seem to help.

I don't know if the problem is with GB or IT but it would be nice if the two could automatically reconnect in this scenario if possible.

petterhs, 2021-04-20:
hmm I'm not able to reproduce this issue running same versions. Do you experience the same if you toggle Bluetooth  off/on on the phone?

GithubUser5462, 2021-04-20:
Yes, exact same result.

JF002, 2021-04-20:
This kind of issue is really hard to debug, and is probably caused by a mix of PineTime HW, InfiniTime, phone HW and software stack. 
InfiniTime does advertise itself automatically as soon as it's disconnected during 3 minutes. It also restarts advertisting every time the display is switched on. Then, it's up to the companion app to receive these advertising messages and reconnect to the watch.

pfeerick, 2021-04-21:
Not knowing that behaviour is most likely the 'cause' of the problem... i.e. I would probably have had a similar complaint if I didn't know that turning the screen on wakes up advertising. 

> InfiniTime does advertise itself automatically as soon as it's disconnected during 3 minutes. It also restarts advertisting every time the display is switched on. Then, it's up to the companion app to receive these advertising messages and reconnect to the watch.

Maybe a future thought could be have periodic advertising... say once every five minutes or so, for 10 seconds or whatever (I'm sure there's some convention that can be followed - or not!) ... i.e. try and make itself available of the phone wants to reconnect.

GithubUser5462, 2021-04-21:
I made a couple more tries to try to understand what is happening.
It seems that it is not limited to Gadgetbridge since it also happens with nRF Connect.

In nRF Connect when I disable/enable bt the "bonded" tab in nRF Connect gets wiped, but the scanner does detect watch - as not bonded.
If i try to manually connect sometimes it connects, but most of the time it timeouts/errors - logger says `Error 133 (0x85) GATT ERROR`. Which is probably not very helpful.

pfeerick, 2021-04-22:
Whether the device is bonded or not has no impact on the device appearing
in the scanner tab. It is merely a list of all the devices which are
currently averrising, with greyed out entries for those devices that have
recently advtertise, but not presently advertising (Bluetooth advertising
of presence).

Can you try this... Start a scan (in NRF Connect), wait for a few seconds
and confirm the InfiniTime is not active. Then whilst NRF Connect is still
scanning,  press the button on the pinetime momentarily so the screen comes
on. Does it show up then? If so, it is working as intended.

diegosolor, 2021-04-22:
I have the same unstable connection problem with gadgetbridge, only workaround is unpair/repair the pinetime

jjrugui, 2021-05-10:
I can confirm the same issue with reconnection as @diegosolor . Connections gets lost and some times pinetime does not appear in gadgetbridge. Only solution in that case is to restart the watch.

For reference I'm using a Oneplus Nord with A droid 11. I'm a bit lost with how to debug it but let me know if I can do anything to provide help/info about the reconnection issue.

pfeerick, 2021-05-10:
Sometimes I have to tell gadgetbridge to connect twice... I.e. connect, press and hold to disconnect after a few seconds of no activity, and then connect again. With the pinetime screen on to ensure it is actually advertising.

blaueente, 2021-05-12:
Independent of airplane mode: Mine (1.0.0) disconnects sometimes from gadgetbridge, and then does not appear on the nRF scanner, and then can only be reconnected by restarting the pinetime.

jjrugui, 2021-05-13:
I'm having the exact same behaviour as @blaueente . I get multiple disconnects per day and I need to restart the pinetime in order to be able to pair it again.

Anyil89, 2021-05-14:
I am on arch Linux arm danct12s  Image. I’m using it on my pinephone,  when I restart my phone like when I update the Bluetooth says it’s disconnected.  But it was connected before the reboot. (my PineTime) But sometimes rarely it will reconnect itself after a while but 80% of the time I have to restart my watch to make it reconnect or disconnect from amazfish and then reset that back up. For it to register.

jjrugui, 2021-06-17:
Today I upgraded Gadgetbridge to latest version 0.58.0 and so far I had a way more stable connection and automatic reconnect working. I'll give more info after testing it for a couple days but I highly recommend the upgrade.

NCLI, 2021-06-24:
I pretty much have to restart the PineTime every 20 hours to connect. Not very useful like that.

1989gironimo, 2021-07-12:
I can add, that not only activating airplane mode causes this behavior. Disabling bluetooth for a while triggers this issue as well, even on the newest firmware (1.2 at this time) and newest gadgetbridge (0.58.1) on GrapheneOS (RQ3A.210705.001.2021.07.07.19)

I even have to remove it from gadgetbridge first, restart the phone AND the Pinetime to get a working connection, so this might be somewhat related, but not the same issue..

raananelefant, 2021-07-22:
I can report that upon Airplane mode on/off, PineTime, running Infinitime FW v1.1.0 w/Hebrew support, cannot reconnect automatically/manually using latest Gadgetbridge (v0.58.2).

Workaround is "disconnecting" watch via GB, re-pairing via Android settings (PineTime disappears from "Previously connected devices") and then connecting via GB.

Phone is a Planet Computer Cosmo Communicator running Android v9, Build number Cosmo-9.0-Planet-02032021-V25, Custom build version alps-mpp-p0.mp1-V5.313

Roterfux, 2021-07-25:
Same issue with firmware 1.3.0

aemkai, 2021-07-27:
I have the same problem. 
Watch doesn't reconnect to Gadgetbridge or nRF Connect and also could not be found by BL-Scan of phone. BL-Symbol on watch is off.
Cycle Bluetooth connection or reset of phone doesn't help.
After reset of watch connection is established automatically and without problems with both apps.
Issue appears @ about 20 hours runtime.
Firmware tested with V1.2.0, self compiled V.1.2.0 and self compiled V1.3.0

tealeg, 2021-07-28:
Another one here.  I just received two PineTime's, the current firmware is v1.2.0.  The behaviour is identical on both devices, and with two different Android phones running Gadgetbridge. 

My experience is as per #527 - there's no need for airplane mode, or any shenanigans on the phone, at some point, when the watch is idle, it will disconnect and will no longer advertise until a restart is forced.   As others have noted, the Bluetooth indication on the watch face disappears, and (naively) I assume the Bluetooth radio has been powered down.

jamesstott, 2021-07-28:
Same problem here as well with v1.3.0 an Gadgetbridge.

1989gironimo, 2021-07-28:
> Another one here. I just received two PineTime's, the current firmware is v1.2.0. The behaviour is identical on both devices, and with two different Android phones running Gadgetbridge.
>
> My [...]

Well, in my case (I've tested it already several times) it is NOT sufficient just to restart the watch.
I've to switch off bluetooth and switch it on again (instead of rebooting, that's a little faster now :) ), because Gadgetbridge cannot scan for devices correctly. If I don't do this, it just hangs and gets unusable.

But the watch also gets disconnected after some time, as someone described above and doesn't reconnect anymore after approx. 20h.

JackofSpades707, 2021-08-08:
I just wanted to chime in and say, if I reboot my phone, it won't connect to my pinetime without unpairing and repairing with GadgetBridge. (stuck on waiting for reconnect)
I suspect the same will be true if I toggle airplane mode or disable/enable Bluetooth.

PineTime: 
- v1.2.0

Phone:
- OnePlus 6T (Sprint model) - A6013
- OxygenOS 10.3.10 (rooted, no modules mucking with bluetooth)

GadgetBridge: 
- version: 0.58.2
- CompanionDevice pairing: enabled

bearclaws8, 2021-08-13:
I can confirm I have the same problem with a new PineTime. Just like @heinrich-ulbricht I received watch, updated firmware from 1.2.0 to 1.3.0, paired with Gadgetbridge, disabled Bluetooth, enabled Bluetooth... no connection anymore.

I have had some better connectivity with nRF Connect compared to GB.  GB seems to dislike reconnecting when toggling Bluetooth while nRF will usually reconnect.  However, after a while, like noted in issue #527 the PineTime itself will no longer broadcast via BLE.

Might it be that this issue and #527 are connected but due to two different issues, one that GB is a flaky and the other where PineTime stops broadcasting?

Is it possible to add into the settings the ability to turn Bluetooth on/off and force the service to restart without restarting the entire device?  On/off may be nice for those looking to maximize battery and security and restarting the BLE broadcast may help with debugging.

Avamander, 2021-08-13:
> GB seems to dislike reconnecting

It doesn't if the disconnection is too long, primarily because making a new connection itself is very unreliable right now. Once InfiniTime can keep a connection in-range and has working reconnects, Gadgetbridge's auto-reconnection can be dialed up.

bearclaws8, 2021-08-13:
> It doesn't if the disconnection is too long, primarily because making a new connection itself is very unreliable right now. Once InfiniTime can keep a connection in-range and has working reconnects, Gadgetbridge's auto-reconnection can be dialed up.

@Avamander  thank you for the clarification.

heinrich-ulbricht, 2021-08-16:
This issue seems to be a rather complex one due to the number of potentially involved parts (and parties):

- Gadgetbridge might have an issue with reconnecting
- Infinitime might have an issue with reconnecting
- the watches hardware might have an issue with reconnecting (?)
- maybe the phone or its operating system plays a role, too?

Furthermore only some users seem affected. (Otherwise this would have priority over new features? Can we estimate how many users are affected?) So the issue might prove hard to reproduce for those who are capable of analyzing and/or fixing this issue.

From an end user's point of view it's frustrating. The connection between watch and phone is unusable for me. Going out of range often is enough to kill the Bluetooth connection. I then have to re-pair, which takes time and after some time I don't bother anymore. I also see that this issue has been known a long time already which gives me the fear it will never be fixed.

From a developer's point of view I'd like to help get this sorted out. I just don't know how and somebody would have to tell me what to do to diagnose this. Reading logs with adb? Can do. Fire up a dev environment? Not so sure. What else?

How is the current state for this issue? Is somebody currently working on it? Can we assume that this will sort itself out with future updates to software and/or hardware?

Back in the Windows Phone times I used to write: I really want this to work. Seems like I'm back again writing: I really, really, reeeeeally want this to work!!

Cuperino, 2021-08-17:
My PineTime is having similar issues. It won't reconnect after a few minutes. Nevertheless, existing connections can last for hours. I have not yet checked what's the longest it will last.

Upon resetting I sync to both my laptop, using Siglo, and my stock Android phone, using GadgetBridge, one after the other before the PineTime's screen turns off. After disconnecting, the paired device needs to be removed from GadgetBridge and the PineTime needs to be reset for it to connect to either device.

PineTime:

    FW: v1.3.0 and v1.2.0
    HW: v1.0.0

Phone:

    Google Pixel 3a
    Android 11 (stock)

GadgetBridge:

    version: 0.59.0
    CompanionDevice pairing: enabled

Laptop:

    Dell XPS 13 7390 2-in-1
    KDE neon Testing Edition
    Linux Kernel 5.11.0-27-lowlatency

Siglo:

    version: Flathub stable released on August, 04 2021

Debo-rah, 2021-08-17:
The same here. Sometimes it helped to reinstall gadgetbridge, sometimes it helped to reboot the watch. After updating to firmware v.1.3 it seemed a little better, but then I got the same problem as someone mentioned above, that it got stuck at the infinitime logo, even though I DID validate the new fimware.

`
FW: v1.3.0 and v.1.2.0
HW: v1.0.0`

Phone:
`Google Pixel 2
Android 10 (Lineage 17.1)`

GadgetBridge:
`version: 0.59.0`

heinrich-ulbricht, 2021-08-18:
@aemkai I read this, too, but this seems like a second problem. For me I 100% can reproduce non-connectivity by disabling+enabling my phone's Bluetooth adapter. No connection afterwards. Fixed by removing+re-adding the watch from Gadgetbridge.

As for reconnecting after being out of range: this seems to work. I fooled myself once or twice when I disabled location services, then went out of Bluetooth range and wondered why it didn't reconnect after coming into range again. Seems like location needs to be enabled for this to succeed.

aemkai, 2021-08-18:
@heinrich-ulbricht 
i can disable and re-enable without problems. The bluetooth icon at the watch also disappears and vice versa.
Location is disabled most time without problems.
Maybe the issue isn't caused by the watch and should be searched at the app in combination with the OS.

Avamander, 2021-08-18:
@aemkai If you look at this current thread then you can see how the issue isn't OS or companion-specific.

aemkai, 2021-08-18:
@Avamander 
Why?
Some users say, a restart of watch isn't sufficient. Others must remove the watch from Gadgetbridge and add it again. This doesn't seem to be an issue of the watch (alone).
The next point is, why some users have problems, others not. An explanation could be the difference between pre- and self-compiled firmware, but this doesn't mitigate the other points.

Avamander, 2021-08-18:
@aemkai I'm not sure what's unclear? The issue exists across multiple OSs and companions (which can also handle other watches great), it's unlikely to be specific to those.

aemkai, 2021-08-18:
@Avamander This doesn't explain my considerations. Also I wrote "app in combination with the OS". Not every app has the same behaviour even on different builds of the same OS. Then e.q. energy saving or privacy settings have different behaviours.
If OS/companion isn't the issue, why isn't a reset of the watch the solution? 
For example @Debo-rah wrote, she had to reinstall Gadgetbridge. How should this need be caused by the watch?

Avamander, 2021-08-18:
@aemkai
> why isn't a reset of the watch the solution?

Because Bluetooth stacks can make assumptions when bonded, that InfiniTime violates.

> How should this need be caused by the watch?

It wasn't, the reinstall was very likely unnecessary.

aemkai, 2021-08-19:
@Avamander I played arond a little bit with the 2 watches, changed airplane mode, bluetooth dis-/enable, (dis-)bonding etc. and got it (I dont know how exactly) to an point, where a connection to both watches wasn't possible, neither with Gadgetbridge nor with nRF Connect.
I reset one watch, no improvement. I reset the phone, and afterwards both watches (also the non-reset) connected without problems.
Maybe there is an issue at Pinetime - i won't exclude that - but i think, there is also a problem in interaction of different components.

tmilburn, 2021-08-19:
I get the feeling there is an integer overflow bug somewhere. At a 32768Hz clock rate a 32 bit signed integer overflows after just over 18 hours. I have definitely seen the Bluetooth inactive when checked at 19h30m of uptime and other commenters have mentioned 20 hours.

bearclaws8, 2021-08-20:
@tmilburn could an integer overflow cause other devices to refuse to see/connect to the watch after it is reset?  I have noticed that after a few instances of PT dropping Bluetooth, and me restarting the watch, Android will begin to not even see it via Bluetooth without the phone being restarted.  Neither OS Bluetooth settings, Gadgetbridge, nor nRF Connect will pick it up.  Granted this is likely an Android issue/limitation but wasn't sure if it was a behavior that may indicate something the watch was doing.

Disregard if this is not helpful.

heinrich-ulbricht, 2021-08-21:
Today I may have experienced a combination of both issues. The watch did not connect. I removed it from Gadgetbridge and tried to add it again. Usually this helps. But not this time as I also had to reset the watch for Gadgetbridge to find it. This might be the ~20 hour issue.

I wonder how this issues can be tackled. Solving it needs a combination of people able to 

1. reproduce the issue(s), 
2. provide data and logs to analyze, 
3. locate the error(s) and 
4. finally fix them.

We definitely have folks here who can reproduce the issues. I also suspect we've got folks to fix errors. But points 2 and 3 might be difficult. Does somebody know which logs we could check to get useful hints? I'll try an `adb logcat` the coming week. Who knows if something comes up. Unfortunately I have no dev kit at hand so I cannot look into watch logs (I suppose).

bryantqo, 2021-08-23:
Happening daily for me as well. Restart watch, delete from GB, re-pair

tmilburn, 2021-08-23:
@heinrich-ulbricht 

1. Reproducing the issue is relatively easy, keep the watching running for 20 hours. There could be a way to speed this up by starting timers with an initial value close to the signed int rollover.
2. Android logs might be helpful in showing if the issue was whilst connected or whilst advertising
3. I have been looking through the bluetooth stack at all the timers to see if there are any issues with them and so far no luck in finding the error.

tealeg, 2021-08-23:
20 hours is a read herring or a different issue, mine does it sometimes in
less than one hour.

pbryan, 2021-08-23:
> 903918173>,

I don't think it's a red herring. When phone and watch are left in
proximity, I see ~20 hours consistently.

bryantqo, 2021-08-23:
Could it not be both? Regardless there is a disconnect and it doesn't want to reconnect. Another thing I can test is going out of range to see if it reconnects.

bearclaws8, 2021-08-23:
I can confirm that on my setup it will reconnect after separation. I am frequently away from my phone for several hours and my watch reconnects without issue (within the 20 hours).  However, I have had difficulty with multiple reconnects.

To me it seems the duration of disconnect due to separation does not matter (within the 20 hours) but disconnecting too frequently has sometimes shortened the "clock" for the watch not reconnecting. This usually happens when I'm running around so I haven't been able to pay enough attention to reproduce reliably.

I used to have more frequent issues like @tealeg but haven't had that happen much lately. Usually the shorter time for not reconnecting was due to me coming into and out of range very frequently over that timeframe.

tmilburn, 2021-08-24:
I managed to run a test to confirm that one of the issues is that the watch does not advertise when the uptime has surpassed 18 hours 12 minutes (65536 seconds or 2147483648 (2^31) ticks). I connected to the watch at 18 hours successfully and checked that the connection was still up at 18:13. At this point I manually disconnected and attempted to reconnect. This failed and I subsequently couldn't detect any advertisement packets from the watch. 

There must be an integer overflow somewhere causing this problem but I haven't managed to track it down yet. Nor do I have an open unit which I can attach a debugger to.

pbryan, 2021-08-24:
Nice detective work!

aemkai, 2021-08-25:
> I managed to run a test to confirm that one of the issues is that the watch does not advertise when the uptime has surpassed 18 hours 12 minutes (65536 seconds or 2147483648 (2^31) ticks).

How did you calcualte the 2^31 ticks?
The tick frequency should be 1024 Hz (see DateTimeController.cpp and FreeRTOSConfig.h), so 65536 seconds are 67.108.864 ticks (= 2^26). Thus, in my eyes an overflow of a 16-bit-variable should be more likely

bellegarde-c, 2021-09-01:
Found this in my LineageOS logcat logs:

```
09-01 18:00:45.854  2467  3171 E bt_btif : bte_scan_filt_param_cfg_evt, 23
09-01 18:01:04.123  2467  3171 E bt_btm  : Ignoring RNR as the state is not BTM_SEC_STATE_GETTING_NAME
09-01 18:01:09.286  2467  3171 E bt_btm  : Ignoring RNR as the state is not BTM_SEC_STATE_GETTING_NAME
09-01 18:01:09.289  6476  6476 E nodomain.freeyourgadget.gadgetbridge.activities.DiscoveryActivity: Problem with permissions, returning

```

lgs11, 2021-09-04:
PineTime 1.3.0
Gadgetbridge 0.59.2
/e/OS Q-0.18 (based on LineageOS)

Mine disconnects any time the phone is too far from the watch and doesn't reconnect again.

What works for me is to press "Connect new device," select the watch, and then press "Do not pair."

lgs11, 2021-09-14:
Maybe it's just me, but reconnecting seems to work much better now after upgrading GadgetBridge to 0.59.3. Most of the time, I just have to wake the watch to cause a reconnect.

bellegarde-c, 2021-09-14:
Works better now except if I put my phone in airplane mode.

bryantqo, 2021-09-15:
Updated to v1.4.0. At some point overnight it lost connection and is no longer advertising. Using nRF Connect the device is not advertising and cannot connect if previously bonded.

M4xTheMan, 2021-09-16:
I'm not able to connect to my PineTime using Gadgetbridge after I've turned off/on bluetooth or disable airplane mode.

I've to delete the device in gadget bridge and pair it again. Therefore I've to do it everday...

phone OS: Lineage OS 18.1
Android version: 11
Gadgetbridge: 0.59.3
Infinitime: 1.4.0

ainola, 2021-09-17:
Looking through the thread shows nearly all users on LineageOS. I just to confirm that an xperia xz2 with (woefully out of date) stock firmware on Android 10 has this exact issue.

tealeg, 2021-09-19:
I'm using a Xiaomi Redmi Note 7 with MIUI 12.03 on Androd 10
QKQ`.190910.002 - if that's useful to anyone.  Clearly not a LineageOS
specific issue.

akutuzov, 2021-09-21:
I am experiencing the same on Sony Xperia 10 with Sailfish OS 4.2.
It does not have anything to do with applications (I am using AmazFish): the phone itself at some point stops re-connecting to the watch. As previous commentators mentioned, it happens after random time intervals, usually an hour or so, but sometimes a day.

Interestingly, the phone sees the watch via Bluetooth (so it _is_ advertising), but upon an attempt to re-connect, it fails with "Problem with pairing authentication. The passcodes did not match". At this point, only resetting the watch helps.

grassygnoll, 2021-09-24:
New PineTime(r) - Just chiming in with another data point; Pixel 3a (_Android 11_); Gadgetbridge 0.59.3; and sealed PineTime (_InfiniTime 1.4; Build date Sep. 2, 2021 18:36:27_).

Received my watch a few days ago and installed necessary app on phone, then Paired/Bonded and was good to go for about 7-8 hours and then noticed Gadgetbridge no longer was connected to watch.  Have not been able to reconnect since (_have yet to restart/reset the watch but did do all the toggling on the app, Bluetooth, phone restart, etc.  Removed the watch from the Gadgetbridge list of devices and it cannot find it anymore to pair with.  Will restart the watch and (_based on this issue log_) suspect it will reconnect for a short while.

danielgjackson, 2021-09-25:
(Apologies that there's a lot of text -- but hopefully it can lead someone in the right direction.)

I am also seeing the advertising consistently stop after 18.2 hours.  The application firmware continues as normal: successfully requests to start advertising for 2000ms, is notified by the BLE stack that it has stopped after 2000ms, and it starts it again -- no errors seem to surface to the application.  

For reference, `NRF_RTC0` *is* at 32768 Hz, and it is the timer used by *mynewt* (`OS_CPUTIME_TIMER_NUM=5`).  The `systickCounter` uses `NRF_RTC1` with a prescaler to bring it down to 1024 Hz.

The `NRF_RTC0` timer is used to provide `os_cputime_get32()` (through `hal_timer_read()` then `hal_timer_read_bsptimer()`).  It is also used in calculations to set a compare register used to fire an interrupt at a specific time in the future.

The timer is 24-bits and wraps every 512 seconds.  A 32-bit timer is required at a higher level.  For this, the top 8-bits are kept in a variable as the top byte of a 32-bit value (OR'd with the counter when the time is read) and incremented on overflow -- this is referred to as `bsptimer->tmr_cntr`.

When I experimentally initialize this top byte to 0x7F, my device stops advertising after just one 8m32s (512 second) overflow period.  You should be able to reproduce this by modifying `src/libs/mynewt-nimble/porting/nimble/src/hal_timer.c`, at line 472, after `bsptimer->tmr_rtc = 1;`, insert:

```c
bsptimer->tmr_cntr = (uint_32)0x7f << 24;  // HACK: Initialize the hal_timer to overflow, when treated as a SIGNED 32-bit value, after 512 seconds (8m32s).
```

In addition, when I initialize this top byte to 0x7E (one value less), my device stops advertising after twice the overflow period (17m04s, or 1024 seconds).  

Something about this rollover (into the top bit) seems to be causing the advertising to stop.  Importantly: the full 32-bits are not overflowing, but this value would now be negative if it was ever treated as a signed 32-bit value.  I don't believe it is directly, but it may well be treated as signed when a delta value is computed as the difference between two times, and perhaps some logic involving this is incorrect.  

In the interrupt handler `hal_rtc_timer_irq_handler()` (via `nrf52_timer5_irq_handler()`, via `rtc0_isr_addr`, via `isr_rtc0()`), typically a few *compare* events happen every second, and the *overflow* happens every 512 seconds.  After the overflows cause `tmr_rtc` to rollover into the top bit, the compare events suddenly stop.  The timer is left running, and the overflow events still happen. 

The reason the compare events stop firing is that the compare value `rtctimer->EVENTS_COMPARE[NRF_RTC_TIMER_CC_INT]` seems to be around 0x00800000 away (modulo 0x00ffffff) from the current time values.  Such a value is set in `nrf_timer_set_ocmp()` line 178: the compare value to the current counts plus (1<<23) -- but only if the `delta_t` >= (1<<24).  (The `RTC0` compare register itself seems to be -- sensibly -- only 24-bit, and this value isn't masked.)

`nrf_timer_set_ocmp()` is called to deal with the `hal_timer_q` queue entries -- populated from `hal_timer_start_at()` via `os_cputime_timer_start()`, and that seems to be mainly called by `ble_ll_sched_*()` functions -- e.g. `ble_ll_sched_adv_new()`.  

So it is presumably that these compare events are no longer firing (seemingly because of their incorrect `delta_t` values) which affects the logic in the link-layer and prevents the advertising packets from being sent properly.

However, I can't yet see how/why the `delta_t` is negative/large -- I don't have my board soldered up to the debugger, and I can't keep a temporary connection for multiple minutes!  If this is a problem in the stack (and not something in the application/config), I wonder why this wouldn't have been noticed upstream on NimBLE before (although I notice the in-tree version of NimBLE is not that current).

M4xTheMan, 2021-09-25:
Could it be that we're seeing two errors here?

1. Pinetime stops advertising after x hours and has to be restarted
2. Gadgetbridge is not able to reconnect with Pinetime, the pairing has to be removed and the watch has to be paired again

[29 more replies]

PR #796 (2021-12-09): Connect and bond with a passkey
Add Feature - Connect and bond with a passkey

- [X] Passkey pairing - passkey is displayed on watch and entered on phone
- [X] Swipe down to clear passkey screen
- [X] Connection encryption
- [X] Connection bonding
- [X] Automatic reconnects to bonded peripheral\*
- [X] Trusted device on Android
- [X] Persist bond between reboots

\* ~~**Rebooting the watch will cause reconnect failures. You must delete the bond from the phone to reconnect/pair.** Persisting the bond between reboots will resolve this.~~

Closes #224 
Closes #302 
Closes #502 

There are several [... cut]
closed 2021-12-05 by Avamander