InfiniTimeOrg/InfiniTime #303: Stopwatch resets when leaving the app.

PizzaLovingNerd, 2021-04-20:
Hello, I was using the stopwatch, and I noticed that whenever I closed the app, the stop watch would reset back to 0. Here is a video demonstrating the issue: https://www.youtube.com/watch?v=AcRQ5m9-PVQ

The expected behavior is having the stop watch still be running or atleast paused when the app is relaunched.

qbinary, 2021-04-21:
I think the problem is that the event which initializes the app is also called to reset the clock application after pressing the stop button. So that could be fixed by adding an additional method which gets invoked by the initialize event if the state of the stopwatch is not set. Otherwise the old state is kept and the app just continues running with the current state.

Soundtoxin, 2022-01-25:
I got hit by this badly earlier. I started downloading the new LineageOS Nightly OTA to flash and then started a stopwatch. It notified me about the download being done a few seconds after the stopwatch had started which just cleared it.

Itai-Nelken, 2022-01-25:
#783 implements persistence.

Eve1374, 2024-10-30:
Hello, 
I noticed that when receiving a notification while stopwatch is running, it also resets the stopwatch. I think issues are related and so I post it there instead of making a new issue. Tested on commit 00769625881e318e6fe29e926c7bd944c5f049bd

mark9064, 2024-10-30:
You're right about it being related, #2141 should resolve this and is looking to be in really good shape to me