InfiniTimeOrg/InfiniTime #306: Enhancement: Battery app line fills too slow for my taste

qbinary, 2021-04-21:
I would like to see that the line fills much faster. I would propose a maximum time of a second until the bar should be filled completely.

Maybe one could add a `sin` function based animation so that the bar filling process is accelerated in the beginning and decelerated at the end.

pfeerick, 2021-04-21:
I would like that also, or perhaps for it to not to animate at all... I don't really see any point in it doing so, and it detracts from it's purpose as an alternate representation of percentage.

qbinary, 2021-04-21:
In my opinion the functionality should be more important than the UI but an aesthetic UI gives the user a better experience using it. So I think the animation should stay but it definitely needs to be accelerated.

pfeerick, 2021-04-21:
> but an aesthetic UI gives the user a better experience using it

I agree. But at the same time, wonder how the progress bar animation adds value to the user experience. It's not like it's an animation showing something filling or emptying. It's not like the animation at Windows applies to it's progress bars - which is in place... it animates the skin of the bar, not the bar moving to it's resting place. 

I just feel that a *moving* progress bar indicates something is changing... which it isn't... but I'm happy with whatever goes implemented.

qbinary, 2021-04-21:
Yeah I see your point and I think you are right. But it would be neat to have some kind of charging animation. The animation could just happen if the watch is charging. Otherwise it just appears filled. That would be a subtle indication which has an actual purpose. But the time has to be reduced one way or the other.

JF002, 2021-07-13:
Fixed in #470

closed 2021-07-13 by JF002