InfiniTimeOrg/InfiniTime #307: App idea: Sleep tracking

myxor, 2021-04-21:
### New app idea: Sleep tracking

**Things to do:**
- Estimate start and end of sleep depending on movement detection (from acceleration sensor). 
     - Data from heart rate sensor and touch screen activity could be considered as well
- Local logging of data for now
- Store and show data on the phone via companion app (Gadgetbridge has support for this)

He is a write up on how this works in general: https://en.wikipedia.org/wiki/Actigraphy

What do you think?

danipozo, 2021-04-21:
Two things to note:
 - as far as I know, battery life with HR sensor constantly on hasn't been tested,
 - HR sensor processing needs to improve yet, since measurements can be very unstable.

I think this is a cool thing to have.

pfeerick, 2021-04-22:
I would not have thought HR monitoring would be constant when in a sleep mode... more like every 15 minutes to half hour... Plus, if this were being done in sleep mode, unless the accuracy is fixed some other way, it could be by doing a running averaging of the samples over a 30 period, and chucking any outliers that are say 30 above that running average to try and eliminate the spurious readings that sometimes come in. 

Either way, This would be nice to see.. someday soon ;)

MysteriousLog6, 2021-05-04:
I would rather turn on the Heart Rate every five minutes rather than 15Minutes to half an hour...., we could try and calibrate the heart rate readings with a smartwatch we know works. (Someone who has both)

Raupinger, 2021-07-20:
here's a paper on accelerometer based sleep tracking https://www.nature.com/articles/s41598-018-31266-z

MysteriousLog6, 2021-07-21:
> here's a paper on accelerometer based sleep tracking https://www.nature.com/articles/s41598-018-31266-z

Thanks! I will read it and see what we can do.

MysteriousLog6, 2021-07-27:
We could try and have raw accelerometer data with a video of the person wearing it (Creepy I know) so that we can program it to be as accurate as possible , the video could have the person turning on the accelerometer logging (An app for this can be made that logs accelerometer data.) helping us sync the two of them together.

The same person could also wear another smartwatch/fitness tracker (Preferably Apple Watch or Fitbit as they have accurate sleep detection compared to others)

This will give us a small hope , battery life is another concern.

theuser18, 2021-12-22:
Could it potentially interface with Sleep as Android to offload some of the effort required in measuring sleep? [The devs are interested supporting PineTime,](https://forum.urbandroid.org/t/support-for-pinetime/7431) but say the APIs required are not in place yet.

ajack2001my, 2022-01-11:
Any progress with Sleep Tracking on the PineTime?

Avamander, 2022-01-11:
There will be progress when someone picks it up and starts working on it. Until that it waits.