InfiniTimeOrg/InfiniTime #315: Bug: Stopwatch crashes the phone when the analog watchface is selected

eivindm, 2021-04-24:
The OS crashes when exiting the stopwatch app on version 1.0.0. Steps to reproduce:
- Select the analog watchface
- Start stopwatch
- Press pause
- Press stop
- Press the button on the watch.
- Wait a few seconds and the OS restarts the watch

Here is a video showing the steps to reproduce the crash: https://youtu.be/KzSs3Y8W_q8
(I press the button twice at the end of this video, but you only need to press once and wait)

pfeerick, 2021-04-25:
I can reproduce this by following those exact steps. Watch only freezes if you use stop to clear the time, and only with the analog watchface... so nice catch! :)

MFAshby, 2021-05-02:
I just tried it with the digital clock face and got the same. Same steps, just selected the digital clock face first. Firmware 1.0.0.