InfiniTimeOrg/InfiniTime #317: Bootloop after a manual reset while wearing the watch

JF002, 2021-04-27:
Yesterday evening, the battery was completely empty. I put the watch on the charging cradle (it rebooted correctly) and went to sleep.
This morning I took the watch out of the cradle (it was still running fine) and manually reset it (long press on the button). Instead of rebooting into InfiniTime, it entered into an infinite bootloop : Green pine cone, display glitch, pinecone again. I tried to revert to previous firmware and to load the recovery firmware using the bootloader but none would load correctly.

It looks like the bootloader does its work but that InfiniTime cannot freezes during its initialisation and that the watchdog resets the watch. As none of the firmware would run, even the recovery one, it looks like something was preventing the firmware to init. It could be caused by  the I²C bus that is frozen (it happened when we were working on integrating the motion sensor, but we thought we fixed that issue...).

I bootloop'd during more than 7h (8h -> 15h/3PM) and the battery finally drained. I restarted correctly once I charged it again.

I managed to reproduce this issue only once by resetting the watch while wearing it on my wrist. If motion/step sensor involved?

pfeerick, 2021-04-28:
This sounds like the exact same issue I mentioned the discord the other day... I did a manual reset, and it went into an infinite bootloop. I don't believe motion was involved in this case, as it was lying flat on the floor. 

i.e. whilst I was getting ready to video it for #316, I manually reset the watch so that uptime wasn't a factor... and it when spastic... In fact, because of that, I actually have it on video... 

https://youtu.be/JF18_JHMudk

... which ended a couple hours later (probably about 4 hours, as the battery wasn't fully charged...) with this near the end... 

https://youtu.be/MEKfSajl7tk

It started up again fine when I put it in the charge cradle, and has behaved since so far.

GithubUser5462, 2021-04-28:
Mine is looping right now. It was sitting on the table doing nothing for at least 12h.

I tried to reconnect it with the phone but GB and nRF Connect were not able to detect it, even with the scan option and even if I woke up the watch with the button multiple times.

nRF Connect can now detect it before the watch finishes loading but it is stuck in a loop.
Prior firmware and recovery firmware did not fix the loop for me either.

nlfx, 2021-04-29:
I encountered the exact same boot loop two days ago. I managed since then to reproduce twice the problem, but had to restart the watch 100+ times to do it.

However, I didn't find any reliable way to repeat this issue:
- It happened once when wearing the watch, twice with the watch flat on a table.
- It happened when rebooting from time face main menu, and also from battery menu.
- It always happened with Bluetooth connected, but given that I only managed to trigger this bug 3 times and had Bluetooth on in ~ 70% of restarts, that might very well be unrelated.

---------

I'm not familiar with nRF52, but I see in the bootloader code that [all peripherals get created and initialized](https://github.com/JF002/pinetime-mcuboot-bootloader/blob/2e1a66e10310f94d3af6e090b736ba84b7dbc177/hw/bsp/nrf52/src/hal_bsp.c#L111). Could this create some kind of deadlock when InfiniTime tries to [re-initialize](https://github.com/JF002/InfiniTime/blob/eedff2c06c292bda8dcb92fc04c67097fe72e22f/src/systemtask/SystemTask.cpp#L86) the same peripherals (SPI for instance)?

Would it be possible to "un-initialize" these peripheral before [jumping from the bootloader to the PineTime firmware](https://github.com/JF002/pinetime-mcuboot-bootloader/blob/2e1a66e10310f94d3af6e090b736ba84b7dbc177/libs/pinetime_boot/src/pinetime_boot.c#L166)? Maybe some unneeded peripherals running are also the cause for the increased power consumption reported in JF002/pinetime-mcuboot-bootloader#8?

But as said above, I've never worked with nRF52 so these are probably some wild uneducated guesses.

Chasarr, 2021-06-04:
Nothing to add here except that the bug is still there on the latest release, (Infinitime 1.1, bootloader 1.0, recovery 0.14.1) but I assume you already knew that. I fully charged my Pinetime over the night, noticed my phone wouldn't connect to it so I restarted it only to be met with a bootloop.

geekbozu, 2021-07-06:
We know it still exists but never hurts to add more confirmation I guess?
Would not connect over Bluetooth while out, Reset it on my wrist -> boot loop. Battery was at ~60% at this time. 
Infinitime 1.2.0 Bootloader 1.0.0

myxor, 2021-07-15:
I was able to reproduce a freeze as follows:
1. device is charging
2. turn screen on while keep it charing
3. try to open main menu (swipe vom left to right)
4. device is frozen

Only thing which i could do in that state was turning it off but now it won't turn on again. I guess i will have to wait until the battery is drained.

Chasarr, 2021-07-15:
I wasn't able to reproduce, unfortunately

Riksu9000, 2021-07-18:
I just got a bootloop after messing with twiMaster. I was working on #492 and tried to make twiMaster wake up only just before reading and sleep right after. If twiMaster is causing the problem, maybe we should remove all `twiMaster.Sleep()` and `twiMaster.Wakeup()` calls, and just run them in `twiMaster.Read()` and `twiMaster.Write()`, making these functions always safe?

>It could be caused by the I²C bus that is frozen (it happened when we were working on integrating the motion sensor, but we thought we fixed that issue...).

Can the bus somehow be reinitialized on each reset? Can the firmware be made to boot even if i2c doesn't work to reflash the firmware? Or is the motion sensor causing the problem, and only resetting it fixes it?

Riksu9000, 2021-07-22:
Actually this sounds very likely to be caused by the bus being frozen. Manual/watchdog reset can happen at any point, so maybe sometimes it will reset while the bus is being used, causing the devices to get confused. This would explain the inconsistency.

Dperrigo18, 2021-07-31:
I don't have much to add but I'm on 1.3.0 and without me doing anything today it rebooted and is now stuck in a bootloop with the pinecone and 1.0.0.
It has been rebooting at random intervals but it usually goes either to the InfiniTime screen in which I apply the firmware or boots back into InfiniTime.

hubmartin, 2021-08-04:
Just my 2cents, when I pressed the back cover on my new unsealed PineTime I've noticed that pressing the button doesn't turn on the screen on. When I released the pressure on the back cover the watch LCD turned on. It was like the watch sensed the button pressed, but continued to work after releasing the pressure. First I thought that there is a short circuit on the button flex PCB so I added capton tape between solder joint and LCD screen. But the behavior was the same. I was thinking that maybe I'm pressing hard on I2C connector to the pulse sensor. I fixed that by removing some foam plastic between pulse meter pcb and battery.

doenietzomoeilijk, 2021-08-09:
I appear to have run into the same issue. Pinetime is on 1.3.0. After noticing it had lost its Bluetooth connection to GadgetBridge, I manually rebooted, and am now admiring a white-to-green pinecone transition, a bit of a wait, a blink, and then it'll reboot again. Tried blue-coning with no visible effect. Tried redconing but the PineTime reboots before GadgetBridge can get a firmware transfer going, it seems. I can re-pair, and after the boot (or at least when the cone is green), GadgetBridge is able to connect for a short while.

I'll leave it to drain and I'll try again...

myxor, 2021-09-27:
My unit does this since a few days:
1. It boots, shows green pinecone
2. Shows blue pinecone
3. Shows red pinecone
4. After a second or two it starts rebooting

Pressing the button seems not to change anything.
I let the battery drain two times already but not yet any success to get it booting correctly.

Gadgetbridge and nrfConnect App are not able to see the device at any time during the boot loop it seems.

Eshanepicfighter, 2021-12-08:
Unfortunately the plague has claimed another victim. I tried to boot the pinetime into recovery mode to flash new firmware

I got an infinitime screen on the watch and flashed new firmware using nRF connect on iOS. I flashed it and it booted into the safe infinitime logo state. Rinse and repeat

Avamander, 2021-12-08:
Which firmware, why was recovery necessary? Crucial details really.

Riksu9000, 2021-12-08:
> I got an infinitime screen on the watch and flashed new firmware using nRF connect on iOS. I flashed it and it booted into the safe infinitime logo state. Rinse and repeat

This is a bit different because you got the issue when flashing, whereas others got this when manually resetting. Could this be a separate issue?

Also sounds like you have an older bootloader. The new bootloader should look like this. Does the older bootloader have known issues?

[image]

JF002, 2021-12-30:
I'm closing this one as it doesn't seem to reproduce on newer version of the bootloader and InfiniTime. Feel free to re-open if necessary.

closed 2021-12-30 by JF002

PhilDevProg, 2022-09-14:
Hey, this just happened to me on version 1.10.

Dperrigo18, 2023-10-04:
I just wanted to add that my pinetime just started bootlooping recently after the battery reached 0.
I don't believe once it reached 0 it turned off, it just started to bootloop. 
It was running the newest version of infinitime.
Also holding the button to get the blue and red pinecones make no change to the watch bootlooping.
This has also happened to my friends watch some time ago.

djpeter84, 2023-11-09:
Have same issue. Watch was working well for months. now stuck in boot loop. using  boot 1.0.0 and  and  InfiniTime 1.13. Has there been a fix for this?

JHarding86, 2023-11-16:
I just powered on my pinetime for the first time in a while. Booted ok, then I restarted it because Bluetooth wasn't working. Now it's stuck at the infini time logo. Nothing I do will make it boot...

FintasticMan, 2023-11-16:
The screen with the InfiniTime logo is the recovery firmware. To get out of it, you need to connect to the device over bluetooth, and flash a firmware update.

You might need to unpair and re-pair for that to work.

djpeter84, 2023-11-16:
> The screen with the InfiniTime logo is the recovery firmware. To get out of it, you need to connect to the device over bluetooth, and flash a firmware update.
> 
> You might need to unpair and re-pair for that to work.
I get this
[image]
[image]

FintasticMan, 2023-11-16:
Hmm strange, does the watch show anything?

djpeter84, 2023-11-19:
> Hmm strange, does the watch show anything?

no, just the green load pine screen