InfiniTimeOrg/InfiniTime #327: Freeze/crash in About menu since 9f15fd1

nlfx, 2021-05-03:
I've had frequent crashes since yesterday merges. Here is how to reliably reproduce the issue:
- Open and close About menu twice
- Open About menu a third time
- Swipe up to try to access screen 2
- Watch freezes and is reset by watchdog a few seconds later

https://user-images.githubusercontent.com/69880003/116891847-b9bb7a80-ac1e-11eb-9d82-254a69eaa988.mp4

I've bisected the problem to commit 9f15fd1820e620e42fb717fb5d692d44a3f0f70d. However given its content, I don't understand how it could create this crash. Maybe it's a problem with some completely unrelated part of the code which only becomes visible after this commit due to some timing or memory position constraint?

Could someone with a dev kit and SWD access please see if they manage to get more info about what is creating the above crash?

In case this bug is memory location dependent and strings with file path in the firmware change the behavior, here is [my version of the above commit](https://github.com/JF002/InfiniTime/files/6415662/infinitime-about-menu-crash.zip).

Thanks a lot for your help!

MysteriousLog6, 2021-05-04:
What exactly have you changed in the commit?
My first response too was "memory allocation maybe?" but there seems to be enough memory for doing a simple vibration.
It might be something else triggering this maybe the driver?

nlfx, 2021-05-05:
> What exactly have you changed in the commit?

No change, I just did a
```
git checkout 9f15fd1820e620e42fb717fb5d692d44a3f0f70d
git submodule update
```
compiled it and flashed it.

> My first response too was "memory allocation maybe?" but there seems to be enough memory for doing a simple vibration.

What is weird, is that the the crash happens on the About menu ([SystemInfo.cpp](https://github.com/JF002/InfiniTime/blob/9f15fd1820e620e42fb717fb5d692d44a3f0f70d/src/displayapp/screens/SystemInfo.cpp)) which has nothing to do with the motor controller change in this commit.

So indeed, I also initially though that the problem might be that the memory for the stack was at its limit and the additional `motorController` argument passed down the call stack was enough to start overwriting other memory region. But the fact that the crash happen only when accessing the About menu for the 3rd time would make no sense if that was the real problem. Unless I'm missing something?

So then I though that it might be a memory leak or memory corruption in the LVGL, and that's why opening the same menu multiple times resulted in a crash. But then it makes no sense that I cannot reproduce this crash on the previous commit (d141888b828d9a47bbcdee86c3c22f1a6d415c56) which contains exactly the same LVGL version and same menus.

So without SWD access, I really don't think I can solve this issue. If no one has time to test this, I'll see if I can re-order a dev kit when they'll be back in stock.

Avamander, 2021-05-05:
@JF002

>  But the fact that the crash happen only when accessing the About menu for the 3rd time would make no sense if that was the real problem.

Potential memory leak?

On another note: Something like Valgrind but for remotely debugged MCUs would be so nice. If **anyone** knows tools like that that work over a gdb remote, please let me know.

MysteriousLog6, 2021-05-05:
Is the vibration motor causing some kind of limit? Not really sure as it happening only around the third time as @nlfx told is weird.

Is it maybe a glitch in the Watchdog?

Avamander, 2021-05-05:
Vibration motor could cause a brownout event when the battery has a very high internal resistance... but someone with that specific watch would have to actually measure and test out this theory.

nlfx, 2021-05-05:
> Is the vibration motor causing some kind of limit?

The crash happens without ever activating the motor. As explained / shown in the above bug description, just opening / closing the *About* menu 3 time and trying to go to its 2nd screen is enough to crash the watch.

> It's really just predicting on tea leaves when the issue isn't reproducible.

Is it not? Did anyone try to reproduce the steps I've described in my bug report with commit 9f15fd1820e620e42fb717fb5d692d44a3f0f70d or any commit more recent? On my watch it reliably crashes 100% of the time if I follow the steps I've described.

MysteriousLog6, 2021-05-05:
> > Is the vibration motor causing some kind of limit?
> 
> The crash happens without ever activating the motor. As explained / shown in the above bug description, just opening / closing the _About_ menu 3 time and trying to go to its 2nd screen is enough to crash the watch.
> 
> > It's really just predicting on tea leaves when the issue isn't reproducible.
> 
> Is it not? Did anyone try to reproduce the steps I've described in my bug report with commit [9f15fd1](https://github.com/JF002/InfiniTime/commit/9f15fd1820e620e42fb717fb5d692d44a3f0f70d) or any commit more recent? On my watch it reliably crashes 100% of the time if I follow the steps I've described.

Oh thanks for explaining , this seems really weird and the fact that it only happens after three tries (On the third one) , can you try this again but try to access the second menu on the fourth try instead of the third?

nlfx, 2021-05-05:
> Oh thanks for explaining , this seems really weird and the fact that it only happens after three tries (On the third one) , can you try this again but try to access the second menu on the fourth try instead of the third?

It also happens on the 4th, 5th, etc.
If the *Screen1* of the *About* menu has been opened 3 times or more, the watch crashes when trying to access the *Screen2*.

MysteriousLog6, 2021-05-05:
> It also happens on the 4th, 5th, etc.
> If the _Screen1_ of the _About_ menu has been opened 3 times or more, the watch crashes when trying to access the _Screen2_.

What intrigued me was that it only happens at the third time.

MysteriousLog6, 2021-05-06:
I am not sure if this will fix the error or not , can you try replacing the "components/motor/MotorController.cpp" file with this one(Modified it a bit , not sure if it will fix it or not):

[MotorController.cpp.txt](https://github.com/JF002/InfiniTime/files/6434806/MotorController.cpp.txt)

Do rename it back to MotorController.cpp from MotorController.cpp.txt

JF002, 2021-05-16:
The behavior you're describing makes me think of a memory fragmentation issue : after a few executions, the memory (from lvgl, freertos,...) is too fragmented and it cannot find a memory segment big enough for its needs.

However, I've just tried to reproduce the issue on the latest commit on develop with no success for now. It doesn't mean that the issue was auto-magically fixed, but it probably "moved" elsewhere in the code and execution paths :/

JF002, 2021-06-06:
I can easily reproduce this issue : open/close SystemInfo 2-3 times, then, scroll to the 2nd screen of displayInfo aaand... crash!

I spent far too much hours trying to figure out the cause of this crash, with no luck :/ It looks like the stack of the Display task gets corrupted when it crashes and falls into HardFault_handler:
[image]

Here's the stacktrace when the screen is displayed correctly:
[image]

I've tried many things : analyze heap, stack and freertos heap/stack usage, disable all the code inside CreateScreen1() and CreateScreen2(), logging, breakpoints,..  but I cannot understand what's happening!

MysteriousLog6, 2021-06-06:
This sounds really weird , what happens of we were to start removing info from the screen?

JF002, 2021-06-06:
I've tried to remove the content of the info screen completely, it would only display a black screen... and it still crashed!

And I have no clue why its hardfaulting that way. Next experiments: run the whole Screen/ScreenList/Settings/SystemInfo classes on a computer to check with memcheck that everything is ok, check IRQ priority in FreeRTOS, disable some code until the problem disappears,...

MysteriousLog6, 2021-06-07:
> I've tried to remove the content of the info screen completely, it would only display a black screen... and it still crashed!
>
> And I have no clue why its hardfaulting that way. Next experiments: run [...]

Well , Good Luck👍

JF002, 2021-06-07:
This makes no sense:

- checkout and build `develop` (79f0fcb07aa80eb70385223272e29f2ba5657bc8) -> No error
- then checkout 9f15fd1 (the commit referenced in this issue) -> No error
- then clean the build folder and try again -> crash

This bug seems to be "moving" on each commit : it's present in 9f15fd1 and ff00873f974cb1400acf7721251167634e60107a (branch move-heap-to-static) but not on current develop (79f0fcb07aa80eb70385223272e29f2ba5657bc8) and this random commit in-between : 13e3463276114dff838fc8fe281754eecfbe9538.

I've identified 2 consecutive commits:
- crash : 8c3b250dbfe17be61462adc8f84760ce9a648e55
- no crash : 13e3463276114dff838fc8fe281754eecfbe9538

All I can say for now is that this issue is **not** caused by compiler optimizations (-O3) because it also happens in -Og.

Soooo... This bug is probably present for some time, but it probably corrupts memory that is sometimes critical, sometimes not.

I still have no idea how to debug this... any help is welcome!

jonvmey, 2021-06-07:
Are any bits set in any of the fault status registers when the problem occurs?

Raupinger, 2021-06-07:
@jonvmey 
CFSR is 0x20000

Itai-Nelken, 2021-06-08:
I'm pretty sure someone already found this, but does this help: https://stackoverflow.com/questions/53253652/debugging-a-hard-fault-in-arm-cortex-m4

JF002, 2021-06-08:
More info from : https://interrupt.memfault.com/blog/cortex-m-fault-debug

```
(gdb) p/a *(uint32_t[8] *)$msp
$2 = {0x20010000, 0x40011008, 0xe000ef30, 0x10 <_lv_gc_clear_roots+16>, 0x20001f74 <systemTask+1788>, 0x1113d <xPortStartScheduler+156>, 0x10e2e <vPortStartFirstTask+22>, 0x61000000}

(gdb) p/a *(uint32_t[8] *)$psp
$3 = {0x200080e8 <ucHeap+14016>, 0x2000e4ac, 0x2000db48, 0x2600 <Pinetime::Applications::Screens::SystemInfo::CreateScreen2()+408>, 0x0 <_lv_gc_clear_roots>, 0xec7 <std::function<std::unique_ptr<Pinetime::Applications::Screens::Screen, std::default_delete<Pinetime::Applications::Screens::Screen> > ()>::operator()() const+12>, 0x2600 <Pinetime::Applications::Screens::SystemInfo::CreateScreen2()+408>, 0x0 <_lv_gc_clear_roots>}
```

So, if we look at MSP (Main Stack Pointer):

R0 = 0x20010000, 
R1 = 0x40011008, 
R2 = 0xe000ef30, 
R3 = 0x10 <_lv_gc_clear_roots+16>, 
R12 = 0x20001f74 <systemTask+1788>, 
LR = 0x1113d <xPortStartScheduler+156>, 
PC = 0x10e2e <vPortStartFirstTask+22>, 
xPSR = 0x61000000

-> This is the main stack, where the scheduler of FreeRTOS was called.
Disassembly:
 ```
vPortStartFirstTask():
   10e18:	4805      	ldr	r0, [pc, #20]	; (10e30 <vPortStartFirstTask+0x18>)
   10e1a:	6800      	ldr	r0, [r0, #0]
   10e1c:	f380 8808 	msr	MSP, r0
   10e20:	b662      	cpsie	i
   10e22:	b661      	cpsie	f
   10e24:	f3bf 8f4f 	dsb	sy
   10e28:	f3bf 8f6f 	isb	sy
   10e2c:	df00      	svc	0
   10e2e:	bf00      	nop
   10e30:	00000000 	.word	0x00000000
```
And the coresponding instruction:
```
   10e2e:	bf00      	nop 
```

PSP (Process Stack Pointer) : 

R0 = 0x200080e8 <ucHeap+14016>
R1 = 0x2000e4ac, 
R2 = 0x2000db48, 
R3 = 0x2600 <Pinetime::Applications::Screens::SystemInfo::CreateScreen2()+408>
R12 = 0x0 <_lv_gc_clear_roots>, 
LR = 0xec7 <std::function<std::unique_ptr<Pinetime::Applications::Screens::Screen, std::default_delete<Pinetime::Applications::Screens::Screen> > ()>::operator()() const+12>, 
PC = 0x2600 <Pinetime::Applications::Screens::SystemInfo::CreateScreen2()+408>, 
xPSR = 0x0 <_lv_gc_clear_roots>

According to the map file, 0x2600 is an instruction from `Pinetime::Applications::Screens::SystemInfo::CreateScreen2()` which is the method called to create the 2nd screen of system info, exactly what I was trying to do at that time. Here is the dissassembly:

```
Pinetime::Applications::Screens::SystemInfo::CreateScreen2():
    2468:	e92d 4ff0 	stmdb	sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
    246c:	ed2d 8b02 	vpush	{d8}
    2470:	b099      	sub	sp, #100	; 0x64
    2472:	4681      	mov	r9, r0
    2474:	460c      	mov	r4, r1
    2476:	690d      	ldr	r5, [r1, #16]
_ZNKSt5arrayIiLj5EE4dataEv():
    2478:	1d2b      	adds	r3, r5, #4
_ZNKSt5arrayIiLj5EE3endEv():
    247a:	f105 0018 	add.w	r0, r5, #24
    247e:	2200      	movs	r2, #0
_ZSt10accumulateIPKiiET0_T_S3_S2_():
    2480:	4298      	cmp	r0, r3
    2482:	d003      	beq.n	248c <Pinetime::Applications::Screens::SystemInfo::CreateScreen2()+0x24>
[... 165 lines cut ...]
    2600:	0004c778 	.word	0x0004c778
    2604:	c22e4507 	.word	0xc22e4507
    2608:	00015180 	.word	0x00015180
    260c:	91a2b3c5 	.word	0x91a2b3c5
    2610:	88888889 	.word	0x88888889
    2614:	42c80000 	.word	0x42c80000
    2618:	0004c780 	.word	0x0004c780
    261c:	0004c750 	.word	0x0004c750
    2620:	0004c754 	.word	0x0004c754
    2624:	0004c75c 	.word	0x0004c75c
    2628:	0004c764 	.word	0x0004c764
    262c:	0004c768 	.word	0x0004c768
    2630:	0004c770 	.word	0x0004c770
    2634:	0004c774 	.word	0x0004c774
    2638:	0004c744 	.word	0x0004c744
    263c:	0004c748 	.word	0x0004c748
 ```

And the corresponding instruction : 
```
    2600:	0004c778 	.word	0x0004c778
```
What is it?

In my opinion, it makes more sense to analyze the PSP, as the display task is running at that time and PC correspond exactly to what the firmware was supposed to do at that time. Now, why does this instruction at 0x2600 crashes the CPU?

jonvmey, 2021-06-08:
0x2600 doesn't contain an instruction it contains data, so it makes sense that executing it would cause a fault. The real question is how is 0x2600 getting into the PC?

JF002, 2021-06-09:
> The real question is how is 0x2600 getting into the PC?
The answer might be 42 ;-D

Ok, I've dug into freertos, irq priorities, task priorities,... everything looks good to me.

And then, I looked at the code... for the 1000th time... and here's what I found:
 - `DisplayApp::currentScreen` points to the current screen, allocated on the heap.
 - When Settings is running, `DisplayApp::currentScreen` points to that instance of `Settings`
 - Settings has a list of `List` as member variable
 - When you tap on an entry from this list, `DisplayApp::StartApp()` is called
 - This method destroys `DisplayApp::currentScreen` to create a new app (the one you selected in the menu)
 - Wait... let's have a look at the callstack:
   - `DisplayApp::lv_taskl_handler`
   - ...
   - `List::OnButtonEvent()`
   - `DisplayApp::StartApp()`
   - `DisplayAp::LoadApp()`
   - `currentScreen.reset(nullptr);`
 - When currentScreen is reset, the dtor() of the object (Settings, in this case) is called. The dtor() of Settings will destroy all member variables of Settings, List included! It means that the code is destroying the object that called this method... What happens to the stack in this case? Probably nothing good.

You know what's scary? ApplicationList and Settings work this way since 1.0!

jonvmey, 2021-06-09:
Oh... Yeah, that's bad. Definitely use-after-free issues in some (all?) of the "nested" application constructs.

JF002, 2021-06-09:
Yes, I think so. This pattern is used by ApplicationList and Settings to load an app when the entry is selected. 

We have a design issue here :  we have a unique_ptr AND a raw ptr pointing to the app:
 - DisplayApp maintains the current app in currentScreen (unique_ptr). Ideally, that unique_ptr should be the only ptr to the app.
 - The ptr to `this` that is stored in `user_data` of the button when creating the app. In fact, the app is leaking a ptr to itself in the constructor

In this case, we are calling the app using the raw pointer, it calls DisplayApp which destroys  the app using the unique_ptr... It's "the snake that bites its tail"...

JF002, 2021-06-09:
> Would it make sense for DisplayApp to have something like a stack of applications for situations like this?

That's an idea. I tried to avoid nested app to keep the memory usage under control and to keep the code simple, but we can think about that.

JF002, 2021-06-12:
For those interested, here is a simple project that reproduces the issue that was fixed:

A.h:
```
#pragma once
#include <memory>
#include "B.h"

class A {
  public:
    A() {
      ptr = std::make_unique<B>(this);
    }

    void Test() {
      std::cout << "A::Test" << std::endl;
      ptr->Test();
    }

    void Delete() {
      std::cout << "A::Delete()" << std::endl;
      ptr.reset(nullptr);
    }

  private:
    std::unique_ptr<B> ptr;
};
```

B.h:
```
#pragma once

class A;
class B {
  public:
    B(A* a);

    ~B();

    void Test();

  private:
    A* ptr;
    int tt = 0;
};
```

B.cpp:
```
#include <iostream>
#include "B.h"
#include "A.h"
B::B(A* a) : ptr{a} {

}

void B::Test() {
  std::cout << "B::Test" << std::endl;
  ptr->Delete();

  std::cout << "Is B alive?" << std::endl;
  tt++;
}

B::~B() {
  std::cout << "B::~B() " << std::to_string(tt) << std::endl;
}
```

main.cpp:
```
#include <iostream>
#include "A.h"

int main() {
  std::cout << "Hello, World!" << std::endl;

  A a;
  a.Test();
  return 0;
}
```

Valgrind is not happy with that code:
```
$ valgrind ./untitled1 
==24823== Memcheck, a memory error detector
==24823== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==24823== Using Valgrind-3.17.0 and LibVEX; rerun with -h for copyright info
==24823== Command: ./untitled1
==24823== 
Hello, World!
A::Test
B::Test
A::Delete()
B::~B() 0
Is B alive?
==24823== Invalid read of size 4
==24823==    at 0x109A25: B::Test() (B.cpp:13)
==24823==    by 0x10943D: A::Test() (A.h:13)
==24823==    by 0x109260: main (main.cpp:8)
==24823==  Address 0x4d9c0c8 is 8 bytes inside a block of size 16 free'd
==24823==    at 0x48429AB: operator delete(void*, unsigned long) (vg_replace_malloc.c:814)
==24823==    by 0x1096B4: std::default_delete<B>::operator()(B*) const (unique_ptr.h:85)
==24823==    by 0x109770: std::__uniq_ptr_impl<B, std::default_delete<B> >::reset(B*) (unique_ptr.h:182)
==24823==    by 0x109E10: std::unique_ptr<B, std::default_delete<B> >::reset(B*) (unique_ptr.h:456)
==24823==    by 0x109CF6: A::Delete() (A.h:18)
==24823==    by 0x1099F8: B::Test() (B.cpp:10)
==24823==    by 0x10943D: A::Test() (A.h:13)
==24823==    by 0x109260: main (main.cpp:8)
==24823==  Block was alloc'd at
==24823==    at 0x483FF3F: operator new(unsigned long) (vg_replace_malloc.c:417)
==24823==    by 0x109565: std::_MakeUniq<B>::__single_object std::make_unique<B, A*>(A*&&) (unique_ptr.h:962)
==24823==    by 0x10939E: A::A() (A.h:8)
==24823==    by 0x109254: main (main.cpp:7)
==24823== 
==24823== Invalid write of size 4
==24823==    at 0x109A2F: B::Test() (B.cpp:13)
==24823==    by 0x10943D: A::Test() (A.h:13)
==24823==    by 0x109260: main (main.cpp:8)
==24823==  Address 0x4d9c0c8 is 8 bytes inside a block of size 16 free'd
==24823==    at 0x48429AB: operator delete(void*, unsigned long) (vg_replace_malloc.c:814)
==24823==    by 0x1096B4: std::default_delete<B>::operator()(B*) const (unique_ptr.h:85)
==24823==    by 0x109770: std::__uniq_ptr_impl<B, std::default_delete<B> >::reset(B*) (unique_ptr.h:182)
==24823==    by 0x109E10: std::unique_ptr<B, std::default_delete<B> >::reset(B*) (unique_ptr.h:456)
==24823==    by 0x109CF6: A::Delete() (A.h:18)
==24823==    by 0x1099F8: B::Test() (B.cpp:10)
==24823==    by 0x10943D: A::Test() (A.h:13)
==24823==    by 0x109260: main (main.cpp:8)
==24823==  Block was alloc'd at
==24823==    at 0x483FF3F: operator new(unsigned long) (vg_replace_malloc.c:417)
==24823==    by 0x109565: std::_MakeUniq<B>::__single_object std::make_unique<B, A*>(A*&&) (unique_ptr.h:962)
==24823==    by 0x10939E: A::A() (A.h:8)
==24823==    by 0x109254: main (main.cpp:7)
==24823== 
==24823== 
==24823== HEAP SUMMARY:
==24823==     in use at exit: 0 bytes in 0 blocks
==24823==   total heap usage: 3 allocs, 3 frees, 73,744 bytes allocated
==24823== 
==24823== All heap blocks were freed -- no leaks are possible
==24823== 
==24823== For lists of detected and suppressed errors, rerun with: -s
==24823== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

closed 2021-06-12 by JF002

nlfx, 2021-06-12:
@JF002 Wow, that was some nasty bug, and must have been a pain to troubleshoot. Thank you so much for tracking this down, finding the root cause, and fixing it :-)

Avamander, 2021-06-12:
I'll use the opportunity to reference https://liberapay.com/JF002, there's a nice way to support the project and its lead maintainer.

JF002, 2021-06-12:
> @JF002 Wow, that was some nasty bug, and must have been a pain to troubleshoot. Thank you so much for tracking this down, finding the root cause, and fixing it :-)

Oooh yeah, I spent way too much time on this issue! But I'm happy I found the issue and fixed it! Thanks for your support !