InfiniTimeOrg/InfiniTime #329: Unable to open apps

DylanVanAssche, 2021-05-03:
# Description

I received my sealed PineTime today and flashed the latest bootloader, recovery and InfiniTime (in that order).
Swiping works, double tap to blank the screen works, but tapping on an app doesn't do anything.

I don't suspect a HW failure as double tap works, so I think it's a software bug?

# What's expected?

Apps open when you tap on them.

# What's happening?

Apps don't open when you tap on them.

# Steps to reproduce

1. Swipe up to the app grid
2. Tap on any app
3. Nothing happens

# Environment

- PineTime sealed
- InfiniTime 1.0.0
- Bootloader 0.14.1
- Recovery 0.14.1
- Siglo

DylanVanAssche, 2021-05-03:
https://user-images.githubusercontent.com/4999159/116910552-512bc800-ac35-11eb-9e64-e4f9eb331e8d.mp4

DylanVanAssche, 2021-05-03:
https://user-images.githubusercontent.com/4999159/116912157-5e49b680-ac37-11eb-87cd-2f52b0017144.mp4

JF002, 2021-05-07:
Here's a little update on this issue : I talked with Dylan, we tried many things like downgrading to InfiniTime 0.7.1, drain the battery to force a hardware reset,... But nothing worked, and the touchpanel continued to behave very strangely.

We contacted Pine64 and they'll send a new unit to Dylan.

We really hope this is an isolated issue and that the new unit will work as expected :)

closed 2021-05-07 by JF002