InfiniTimeOrg/InfiniTime #331: Remove the paint app

GIJack, 2021-05-04:
I cannot think of a more useless program on any platform I've used in over 30 years of doing computers
1. it serves precisely zero purpose. There are dozens of better platforms for this kind of thing
2. its only 1 color, white on black background
3. no paint features for making any useful shapes
4. no ability to save or load your painting
5. using touchscreen with your finger is super awkward
6. It consumes storage and memory, which are limited on this platform, and might contribute to OOMs
7. Its another icon to accidentally click on.
8. There are literally no other features besides turning pixels from black to white,

Avamander, 2021-05-04:
I'd consider it as a touch panel test app with visual response. It's also very tiny, nothing significant would be gained by removing it.

GIJack, 2021-05-05:
Indeed. It is in fact a test app, that was absolutely great when infinitime was alpha/beta/development environment.

This should have been removed prior to 1.0.0 release, and really should be removed in any version tagged for production.

deeg05, 2021-05-05:
I think better solution would be moving it into "Settings" as "Touchscreen Test" option

Kozova1, 2021-05-06:
@GIJack for points 2 and 8: The paint app supports multiple colors - touch and hold on the screen to change the color.

GIJack, 2021-05-06:
@deeg05 makes a point. That said I feel the option in settings should simply be "turn test apps on/off", and this should include paint and the two games.

JF002, 2021-05-07:
~~1. Why should everything serve a special purpose? This is a test/demo app, and you can build something on top of it.~~
~~2. You can change the color by long tapping the screen~~
~~3. Do the shape with your finger~~
~~4. You can contribute to this feature~~
~~5. Why do you use a smartwatch is a touchscreen, then?~~
~~6. Did you measure it?~~
~~7. Ok, let's delete all apps so you don't accidentally tap on the wrong one~~
~~8. Draw a heart on it and spread peace in the world~~

One day, one of my teacher I respect a lot told me to never erase something I wrote, even if it looks like a silly mistake. Instead, I should just cross it out. She considered that anything we write could be useful in the future, to remember why we came to a certain conclusion, to not make the same mistake again, or because we might change our mind and maybe that why not a mistake after all...

You might have a lot of experience in computers, but you are very new to this project, and that's probably why you think this app is a silly mistake. This app allowed us to improve the touch and display drivers performance and stability. This app also served as a base for developing other apps. It also allowed us to add the new *continuous* mode for the touch driver. I also used it to build a demo of a [BLE HID mouse running on the PineTime](https://video.codingfield.com/videos/watch/9bb4cd12-50fb-41a3-adb1-4f53a9a6d533).

So, just because you think this app is not useful and because you don't like the games (really?), you want us to delete a part of our history? Just because you this this app is not useful, and because you don't like the games, you want us to miss the opportunity to inspire creative minds who would build awesome things based on these apps?

Instead of saying that this app is not useful and that we should remove it along with the 2 games, why don't you come up with idea to improve them, to leverage what we have done to effectively do something useful or fun? You think these apps should have never reached 1.0? Sorry but too late! Where were you when we were talking about the roadmap of InfiniTime 1.0?

I kindly remind you that I've never said that 1.0 meant that  InfiniTime reached a *perfect* state that would compete with with other commercial products. Quite the opposite! I recommend you read again [the announcement of the release](https://www.pine64.org/2021/04/22/its-time-infinitime-1-0/) so you can better understand the philosophy of this project.

~~Now, please contribute positively to this project of find another project to evangelize with your 3 decades long experience in "doing computers".~~
Thanks for your feedback!

closed 2021-05-07 by JF002