InfiniTimeOrg/InfiniTime #335: Optimizing the build size

joaquimorg, 2021-05-07:
Using the option  "-Os" in the build reduces the size of the build.

The only problem is that it removes all the debug information, but also in builds on the master there is no need to have that information.

[image]

It may be an idea to change the build, since we cannot use the debug, so we could switch to this option and the debug removes the "-Os" and in the release makes the build with this option.

The idea cames from :

[image]

Avamander, 2021-05-12:
I mentioned this under some other issue, but `-Os` is not necessarily fast, hot paths should be built with `-O3`. Ideally other optimizations would be taken before losing out in speed.

JF002, 2021-05-16:
I suggest we test this in a PR in milestone 1.2 and decide if we keep it, or improve it before the release of 1.2

closed 2021-06-01 by JF002