InfiniTimeOrg/InfiniTime #339: FR: Show call notifications when no caller info present

pfeerick, 2021-05-12:
When a private / no caller id call is received, this in no text string information provided (but category and alert number are), and InfiniTime doesn't display the alert. 

This FR proposes that it instead behaves similar to the Pebble, and simply shows the string 'Unknown' and process the new alert accordingly. 

This behaviour can easily be tested in GadgetBridge by simulating an incoming call with an empty caller name via the debug menu.

JF002, 2021-05-16:
This is a tricky one! I had to add [this code](https://github.com/JF002/InfiniTime/blob/develop/src/components/ble/AlertNotificationService.cpp#L57) to ignore notification that contained no data (message size = 0). I don't know what, gadget bridge would send them quite often. 

If we want to receive call notification without data, we would need to revert this check and receive empty notifications again...

I think we should work together with gadgetbridge and maybe other companion app so that they avoid to send empty notification when it's not needed. In this case, we can remove the check for empty notifications.
Another possibility would be to ask them to insert a *dummy* caller ID for private calls so that the notification is not empty when we receive it.

MysteriousLog6, 2021-06-11:
We could also add a specific "type" of notification that the Partner app can use similar to the mi band , thus allowing "Unknown" calls to appear and the app can still send blank notifications.

tomechio, 2022-04-27:
I have solved this problem by adding few lines of code to **notifications.cpp**.

On the line **238** I have replaced:
      lv_label_set_text(alert_caller, msg);

with: 
      if (strcmp(msg, "null") != 0) {
		lv_label_set_text(alert_caller, msg);
	  } else {
		lv_label_set_text(alert_caller, "Unknown");
	  }

Before:
[image]

After:
[image]

JF002, 2022-04-30:
Your fix looks valid, and you could probably open a PR so we can review / test it a bit further.
However, the original post described an issue with call notification that was not displayed at all when no caller ID was provided by the companion app. How can we reproduce a call notification that displays that "null" caller ID?

tomechio, 2022-05-03:
I'm using Gadgetbridge as a companion app and when I receive a call from unknown number(Carrier of the caller has hide it) I receive "null" on my Pinetime.

Few months ago I created an issue for Gadgetbridge but since nothing has happened at the end I decided to add my own code to Infinitime: 
https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2604