InfiniTimeOrg/InfiniTime #340: Bug - Battery Level Characteristic not being updated

pfeerick, 2021-05-12:
The recent GadgetBridge 0.56.2 update adding battery monitoring has brought to light an issue with InfiniTime not updating the Bluetooth Battery Level Characteristic.
* I had an initial value of 59%, and then no updates until the next day with 56%. 
* Disconnecting and re-connecting has had no effect
* NRFConnect is confirming a reading of 56%
* This was with an uptime of about 10 hours
* I'll report back tomorrow when I do my daily reset as to if I can pin down if it only updates once, or updates for a period and then stops

InfiniTime 1.0.0

blaueente, 2021-05-12:
Same here. After resetting (long press) the watch, the value is changed to the current one at the point of resetting.

tmilburn, 2021-08-19:
Is this fixed by e21f6a7? It updates the battery percentage every 10 minutes now

Riksu9000, 2021-08-20:
> Is this fixed by [e21f6a7](https://github.com/JF002/InfiniTime/commit/e21f6a7f414d1f832e8fddfeaab3b9de05aa3459)? It updates the battery percentage every 10 minutes now

There is a bug with that implementation that is fixed by #580.

>Battery is currently only updated when SystemTask receives a message. While the watch is sleeping, it might not receives any messages and the battery level never gets updated. The notification always contains the same percentage. This fixes that as well.

PR #580 (2021-09-12): Make battery reading periodic
Move battery reading to a timer.
Add events for battery notification.
Disable internal pullup because external pullup exists.

Battery is currently only updated when SystemTask receives a message. While the watch is sleeping, it might not receives any messages and the battery level never gets updated. The notification always contains the same percentage. This fixes that as well.

This also keeps the percentage shown in InfiniTime in sync with the one shown in Gadgetbridge better/always*.

Fixes #340
closed 2021-09-12 by JF002