InfiniTimeOrg/InfiniTime #343: "Find My Phone" functionality with Gadgetbridge

Raupinger, 2021-05-13:
I used that feature quite a bit when i was daily driving my Miband. Not sure about what the best place to put it would be though. It doesn't justify being a full watchapp, but the quick settings are full.

JF002, 2021-05-15:
Interesting feature, indeed! I think it would be nice to work together with the Gadgetbridge team to find the best BLE API for this functionality.
Then, this feature could also be added in other companion apps like Amazfish and Siglo :)

cincodenada, 2021-05-15:
This is one of the features I miss most after moving to a FitBit a while back, so I'd love to see this too! As for interface, it would kinda make sense long-pressing the vibration option on the quick settings screen, but that probably shouldn't be the only way to do it at least, since it's not very discoverable. And I'm not sure if long-pressing quick settings is something that's desirable to introduce in the first place.

Raupinger, 2021-05-15:
I don't think we can even detect long presses on individual items

MysteriousLog6, 2021-06-09:
> I used that feature quite a bit when i was daily driving my Miband. Not sure about what the best place to put it would be though. It doesn't justify being a full watchapp, but the quick settings are full.

As a Mi Band User , I cant exaggerate how MUCH I use this , kind of fun too it disturbs everyone!

huckvrni, 2022-03-11:
I also switched to infinitime from the mi band and enjoyed this feature, if I find the time, perhaps I will look into it from the gadgetbridge side...

yusufmte, 2022-05-14:
On 1.9.0 this is working flawlessly for me and appears as a call from GadgetBridge. Anyone still having hiccups with this feature? Or should the issue be closed?

[image]

huckvrni, 2022-05-14:
You got it wrong, what you're showing is 'find my watch', in some other smartbands there is an option to make the phone start ringing and gadgetbridge has support for this.

yusufmte, 2022-05-14:
My bad, misread the title-- you are right. Thank you for the correction.

WhyNotHugo, 2022-05-20:
There's actually a Bluetooth profile for this: https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=239389

I think it would be wise to implement the standard, so rather than require phones to have specific Infinitime support for this, they just need to support the standard.

Also relevant: https://www.bluetooth.com/specifications/specs/find-me-profile-1-0/

devnoname120, 2022-06-20:
No need to rely on this hard-to-implement Bluetooth spec. GadgetBridge already has a device event for that: [GBDeviceEventFindPhone](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/0349b2530fd29c02ad4d7fa85c4144b8cf7c1457/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/deviceevents/GBDeviceEventFindPhone.java#L20-L28)

In order to get it working on InfiniTime, you'd need to:
- Add new UID for a new service in [PineTimeJFConstants.java](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/0349b2530fd29c02ad4d7fa85c4144b8cf7c1457/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/devices/pinetime/PineTimeJFConstants.java#L21): `UUID_SERVICE_INFINITIME`. Alternatively, we could add a service to emulate UART over BLE. See [nRF documentation](6e400001-b5a3-f393-e0a9-e50e24dcca9) and [WASP-OS's implementation in GadgetBridge](https://codeberg.org/Freeyourgadget/Gadgetbridge/src/commit/0349b2530fd29c02ad4d7fa85c4144b8cf7c1457/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/waspos/WaspOSDeviceSupport.java#L215-L233).
- Register this new service.
- Modify `onCharacteristicChanged()` to add support for two new characteristics: `UUID_CHARACTERISTICS_FIND_PHONE_START` and `UUID_CHARACTERISTICS_FIND_PHONE_STOP`. Or just one characteristic with a value defining whether it stops or starts the find device functionality.
- Forward the event to GadgetBridge:

```java
value = ...

GBDeviceEventFindPhone deviceEventFindPhone = new GBDeviceEventFindPhone();
deviceEventFindPhone.event = value ? GBDeviceEventFindPhone.Event.START : GBDeviceEventFindPhone.Event.STOP;
evaluateGBDeviceEvent(deviceEventFindPhone);
```

Then you would need to implement the new device event service + a button or an app in InfiniTime to use this new capability.

WhyNotHugo, 2022-06-20:
> No need to rely on this hard-to-implement Bluetooth spec. GadgetBridge already has a device event for that: [...]

The huge downside of this approach, is that the feature would work with a single application, on a single OS family (Android). I can't imagine anyone willing to implement this custom GadgetBridge API for that on Linux mobile, given that a standard bluetooth spec exists for the exact same purpose.

Following a standard spec means far wider devices support, and more chances of it working on other devices and vendors.

What's the motivation for GadgetBridge to implement their own custom protocol, instead of using the bluetooth standard?

devnoname120, 2022-06-20:
@WhyNotHugo The Bluetooth standard you mention is not available on BLE. Additionally their spec is over-engineered and a pain to implement. It's several orders of magnitude more time-consuming than just implementing it in a few limes in GadgetBridge.

Also just the fact that it's a standard doesn't mean that it would automatically work everywhere. I'd be surprised if it were implemented in the devices you mention.

This said, be my guest if you want to implement it by yourself.

Avamander, 2022-06-20:
> There's actually a Bluetooth profile for this: https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=239389

Wrong Bluetooth.

> I think it would be wise to implement the standard, so rather than require phones to have specific Infinitime support for this, they just need to support the standard.

There's the Immediate Alert service but no OS supports or has that natively. 

> No need to rely on this hard-to-implement Bluetooth spec.

IA is not difficult to implement though.

> In order to get it working on InfiniTime, you'd need to:

It's just as much work to implement IA.

> The huge downside of this approach, is that the feature would work with a single application (GadgetBridge), on a single OS family (Android).

It wouldn't.

> Following a standard spec means far wider devices support, and more chances of it working on other devices and vendors.

As mentioned above, nope.

> What's the motivation for GadgetBridge to implement their own custom protocol, instead of using the bluetooth standard?

They haven't.

> It's several orders of magnitude more time-consuming and a lot more frustration than just implementing it in a few limes in GadgetBridge.

Not really. You're already implementing 98% the same as implementing IA, 1% being the different UUID.

> I'd be surprised if it were implemented in the devices you mention.

It even might be, it is implemented quite often on wearables themselves. The catch is that it rarely works as intended. Doesn't mean IT can't be compliant though.

Avamander, 2022-06-20:
@devnoname120 If you are going to spend time implementing something, it would be very reasonable to actually get acquinted with the actual standard before dismissing it.

Supaiku, 2022-12-27:
@JF002 curious if this is something that might get some attention? I would love to be able to do that with my pimetime!

I see it's already a feature on gadgetbridge for other devices as mentioned (related issue https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2967 )

TechnologyClassroom, 2022-12-27:
I think someone just needs to program a button to do it at this point. That person could be you.

Supaiku, 2022-12-27:
@TechnologyClassroom I wonder if that's true... and how to program a button.

Assuming all that hubbub about bluetooth vs GadgetBridge feature is true it seems using GadgetBridge is probably easier.

It looks like it [equires paring via the Companion Device Management](https://codeberg.org/Freeyourgadget/Gadgetbridge/wiki/Find-phone) to use that feature - anybody know how to find out if Infinitime pairing with GadgetBridge is currently using that?

Supaiku, 2022-12-29:
@TechnologyClassroom while your hypothetical is compelling, I think it would take me a while to learn enough python (i think?) to add a button... I will consider it for my new year's resolution though :P

siick, 2023-08-03:
I am still very interested on this feature even 2 years after i bought my watch :)

WhyNotHugo, 2023-09-25:
This issues is a bit of a chicken-and-egg situation.

An implementation is needed on the watch side (the "**Locator**"), and an implementation is also needed on the phone/laptop side (the "**Target**"). I don't think that it's feasible to test one without the other.

I have written a tiny "Target" implementation using [bluer](https://github.com/bluez/bluer): https://git.sr.ht/~whynothugo/find-my-tux. It works on Linux desktop or mobile (or laptop). For the moment it just prints "Got an alert", but once there's a "Locator" implementation for InfiniTime I can refine this to execute a user-configurable command.

Supporting other platforms (e.g.: BSD) is non-trivial, given that the entire Bluetooth software stack is different. An entirely different implementation is required.

As you might have guessed, this is pretty untested, given the lack of a "Locator"/watch implementation. If someone else wants to pick this up, I'd be very happy to test the InfiniTime changes. These are the relevant specs:

- Find Me Profile 1.0: https://www.bluetooth.com/specifications/specs/find-me-profile-1-0/
- **Immediate Alert Service 1.0: https://www.bluetooth.com/specifications/specs/immediate-alert-service-1-0/**
- [The assigned number for Immediate Alert Service is 0x1802](https://btprodspecificationrefs.blob.core.windows.net/assigned-numbers/Assigned%20Number%20Types/Assigned_Numbers.pdf).

The whole profile is just one service with a single characteristic; it's exceedingly simple. The "Locator" implementation only needs to execute a "Write without Response" with the correct alert level on the "Target".

jmlich, 2023-10-17:
By the way, ImmediateAlertService is already implemented in InfiniTime
https://github.com/InfiniTimeOrg/InfiniTime/blob/main/src/components/ble/ImmediateAlertService.h

Avamander, 2023-10-17:
@jmlich In the wrong direction (that's for Find My PineTime).