InfiniTimeOrg/InfiniTime #344: Timer functionality.

Raupinger, 2021-05-13:
Add a timer, that can be set, triggering a vibration when it runs out. A good way to put this in the UI might be with a left/right swipe on the stopwatch app.

pfeerick, 2021-05-13:
I was sure this had been mentioned in another issue, but perhaps it was only on the chat platform. 

Anyway, I would prefer this gets its own icon, rather than embedded in another app... It makes a little more sense on a phone as you have the screen realestate to have nice big buttons and link it all under a clock app, but I think single task apps suit a watch better.

JF002, 2021-05-15:
That's actually a feature I would like to see in InfiniTime in the near future :)
I agree with @pfeerick , I think this functionality deserves its own app.

Raupinger, 2021-05-20:
PR is merged, I'll close this

closed 2021-05-20 by Raupinger